js4jiang5/BatteryOptimizer_for_Mac #1: 請教關於 battery calibrate

k098765, 2024-10-28:
您好

關於  battery calibrate 說明裡的
〃calibration will be started automatically once macbook lid is open and AC adapter is connected〃

是指每次關機、重新開機、闔上螢幕睡眠後的打開螢幕喚醒使用電腦都會自動進行 電池校正 的功能？
還是只有當預設的 battery schedule 時間到的時候，螢幕闔上或未插電源的情況下才會延遲到打開螢幕及接上電源才做電池校正？

我的情況是每一、兩週會關機到另一個城市的時候，重新開機的情況下是否會直接背景進行電池校正？
或者是會開機的時候仍故定維持在 設定的上限 80%， 直到系統遇到設定的schedule才自動做電池校正？ 

我希望是能每次開機都固定維持在設定的上限80%就好， 不要每次開機都校正一次。

另外，在 AIDente Pro 的情況下，  關機後開機， 還是會依照系統的設定讓電池充到100％，然後再慢慢用到設定的上限，並不再充電。  
這樣對於類似我這樣頻繁長時間跨國而習慣性關機的使用者來說，就覺得沒什麼功能性的幫助。
請問 BatteryOptimizer 是否也會有同樣的情況， 或是能夠 每次開機都固定維持在設定的80%上限？

以及，關於 schedule 能設定最久的週數是 每四週就一定會排程做電池校正，  請問是否能延長至兩、三個月再做一次校正，會不會比較好？ 

感謝您做的好用程式，  讓menu bar上沒有 icon 非常棒👍

備註，我的電腦是 M1 max  （已經不夠max的規格，只能給小寫）

raymao9, 2024-10-28:
以我用過 AlDente 和 bclm 的理解，
`battery maintain 80` 這樣的指令應該並不會在重開機後失效，也不需要在每次開機時重新指定，但如果你習慣關機後還插著電，那程式沒有辦法做到關機狀態下還限制充電上限，我用 AlDente 時也沒有出現你說的「重開機會先充到 100% 再放電到 80%」，你應該是關機後還插電才會這樣，否則本來就應該是「每次開機都固定維持在設定的 80% 上限」，如果你的 AlDente 有加入開機登入項目的話

另外關於電池校正我的理解是
`battery calibrate` 是立即進行電池校正，所以後面的說明是指，如果已經正在執行校正過程中卻蓋上了筆電，那可能會導致校正中斷或是過程中會有通知提醒開蓋，畢竟校正過程通常需要幾個小時甚至大半天，你按了指令過後很可能會忘記，而用腳本指令的方式不像 AlDente 的介面會讓你一眼看出目前是在校正的哪一階段

`battery schedule` 才是排程進行電池校正，所以排程是到預訂進行的指定時間，剛好沒有開蓋的話會跳通知提醒，然後只要在當天 24 小時內有開蓋就會立刻執行（這句在英文中沒看到，是 PTT 文章裡作者講的第 4 點，無論是定時還是立即執行應該都適用）

如果我理解有錯也希望作者糾正~ 謝謝作者

k098765, 2024-10-28:
我是只有在移到其他城市的時候才會關機，收到行李箱， 到飯店再接上電源。    開機狀態也是設定只暗螢幕不關機。   沒有關機還接電源的情況。

我會注意到這個問題是， 接上電源開機後，它還是會照 macOS 內建的電源管理自行排定充滿的時間，   並不是一開機就直接充到100%，   但是它會是參照使用者的習慣背景排定在一、兩天內自動充滿到100%。

而我所預期的是 app限定在 80%，   重開機也只固定電量在 80%，不會再充到滿。

raymao9, 2024-10-28:
> 我是只有在移到其他城市的時候才會關機，收到行李箱， 到飯店再接上電源。 開機狀態也是設定只暗螢幕不關機。 沒有關機還接電源的情況。
> 
> 我會注意到這個問題是， 接上電源開機後，它還是會照 macOS 內建的電源管理自行排定充滿的時間， 並不是一開機就直接充到100%， 但是它會是參照使用者的習慣背景排定在一、兩天內自動充滿到100%。

我猜你會不會是以下情況？
1. 用了第三方充電管理但同時還開著內建的最佳化
2. 每天都做電池校正或排程了每天做，因為只有電池校正會充超過自設上限（充滿）、放電，完成後才會恢復充電限制

我天天用 AlDente 的時間是有點久以前了，但印象中不會這樣，而且我幾乎都沒在做電池校正，也不開航行模式，是看到本專案作者討論才去認真考慮這兩個功能，而在此之前的大半年時間，都在用 bclm，從來沒有過充（超過 80%），只有很偶爾出現低於 80% 卻沒有充進去電的狀況，你看如果不是以上問題的話，試一下 bclm 或是本專案，直接使用下去就知道有沒有同樣問題了~

k098765, 2024-10-28:
我沒裝過任何第三方充電管理的程式，   只有前兩週看到 AIDente的介紹裝來試試看， 才又注意到80%時重開機還是會充滿電的情況，所以就刪了。

目前是有裝這個 Battery Optimizer，   想問清楚是否能達到我預期的直接限定80％就不再充電，  或是又會跟 AiDente 那樣重開機就會再充到滿的情況。

AiDente 的網頁上有說到我的情況，Charging Paused的時候關機，開機就會charging to 100%。

https://apphousekitchen.com/stop-charging-when-powered-off/

js4jiang5, 2024-10-28:
恩，我並沒有試過所有的情況。首先你要先關掉 macOS 本身的電池最佳化以免相衝突，另外你可以幫忙試試看 BatteryOptimizer 是否關機後真的會充到 100%，這跟我的認知不一樣。我試過 Intel Macbook 關機後是停止充電的。

k098765, 2024-10-28:
非常感謝～   我觀察幾天，再來回報～   😀

js4jiang5, 2024-10-29:
不客氣，這是預期的行為，不過如果你已經更新到 15.1 Sequoia 後，關機後上限就沒作用了，蘋果從 Sequoia 後把這個功能拿掉，所以必須要開機才行。Sonoma 就沒有這個問題，睡眠或關機上限依然有作用。

> 非常感謝～ 我觀察幾天，再來回報～ 😀
> 
> ====
> 
> 趁今天剛好更新 15.1 ， 重開機了三次來測試，都是按照我平常的習慣， 放在桌上後先開機 （今天省去了從行李箱拿出來放到桌上這步驟，我想這個也不會影響到充電） 開機時（apple logo畫面） 裝上充電器 ，這時充電線的燈號顯示橘色：充電中 從登入帳密畫面到顯示桌面， 充電線燈號仍為橘色，menu bar也顯示充電中，電量也確實有上升 不過，在進入桌面後約 1分鐘 左右的時間， 充電狀態會改成 〃not charging〃，充電線燈號改為綠燈。 也確實未再繼續充電。
> 
> ＝ 這是我預期的電源模式～
> 
> 非常感謝作者， 這個程式太棒了～ 👍

closed 2024-12-07 by js4jiang5