js4jiang5/BatteryOptimizer_for_Mac #2: Calibration failed

sdeviljr, 2024-10-28:
**What is the issue? (required)**
battery calibrate discharging completed at 15% and error happened.
It shows
"Calibration Error: Discharge to 15% fail
Recover in 5 secs, wait ...
Battery maintain is recovered"
I didn't know it will keep doing its job or I need to calibrate again so I tried "battery calibrate" again but 
it also failed this time.
I tried to calibrate third time while playing World of warcraft, my MacBook Pro M1 max 14' 2021 suddenly shutdown.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. battery calibrate

**Expected behavior (required)**
Discharge to 15%, charge to 100% and keep it an hour, than back to my % setting(75%)

**Screenshots (optional)**
None.

**Error logs**
None.

**Additional context (optional)**

js4jiang5, 2024-10-28:
@sdeviljr It seems that the discharge process was interrupted and exit, meaning that process was killed for some reason. What is the real battery percentage when the error occurred? Had it ever started discharge? Could you tried to calibrate without playing the game and see what happened?

sdeviljr, 2024-10-28:
The second time calibrated without playing game but still failed.

js4jiang5, 2024-10-29:
Please update to the latest verion using command "battery update" in Terminal.
If the error still occurs, please let me know what percentage the battery level is when the error message shows up.

closed 2024-11-07 by js4jiang5