js4jiang5/BatteryOptimizer_for_Mac #4: 使用出現幾個 error

Lawlietr, 2024-10-29:
使用出現幾個 error
主要是 “Error: SMCWriteKey() = e00002bc”, 跟最後的 "10/29/24-16:24:02 - Error: macbook lid must be open before discharge
10/29/24-16:24:02 - Calibration Error: Discharge to 75% fail"

`~ ❯ battery calibrate
10/29/24-11:21:36 - Calibration: Please open macbook lid and connect AC to start calibration
10/29/24-11:21:46 - Calibration: Calibration has started! Start discharging to 15%
10/29/24-11:21:47 - Suspend in 5 secs, wait ..10/29/24-11:21:48 - Battery maintain is suspended
10/29/24-11:21:48 - Discharging to 15% from 79%
10/29/24-11:21:48 - 🔽🪫 Enabling battery discharging
10/29/24-11:21:48 - 🔌🪫 Disabling battery charging
Error: SMCWriteKey() = e00002bc
10/29/24-11:21:51 - MagSafe LED function invoked
10/29/24-11:21:51 - 💡 Setting magsafe color to none
10/29/24-11:21:51 - setting LED to none
Error: SMCWriteKey() = e00002bc
10/29/24-11:21:51 - Battery at 79% (target 15%)
10/29/24-11:22:51 - Battery at 79% (target 15%)
10/29/24-11:23:51 - Battery at 79% (target 15%)
10/29/24-11:24:51 - Battery at 79% (target 15%)
10/29/24-11:25:51 - Battery at 79% (target 15%)
10/29/24-11:26:51 - Battery at 79% (target 15%)
10/29/24-11:27:51 - Battery at 78% (target 15%)
10/29/24-11:28:51 - Battery at 78% (target 15%)
10/29/24-11:29:51 - Battery at 78% (target 15%)
10/29/24-11:30:51 - Battery at 77% (target 15%)
10/29/24-11:31:51 - Battery at 77% (target 15%)
10/29/24-11:32:52 - Battery at 76% (target 15%)
10/29/24-11:33:52 - Battery at 76% (target 15%)
10/29/24-11:34:52 - Battery at 75% (target 15%)
10/29/24-11:35:52 - Battery at 74% (target 15%)
10/29/24-11:36:52 - Battery at 74% (target 15%)
10/29/24-11:37:52 - Battery at 73% (target 15%)
10/29/24-11:38:52 - Battery at 73% (target 15%)
10/29/24-11:39:52 - Battery at 72% (target 15%)
10/29/24-11:40:52 - Battery at 72% (target 15%)
10/29/24-11:41:52 - Battery at 71% (target 15%)
10/29/24-11:42:52 - Battery at 70% (target 15%)
10/29/24-11:43:52 - Battery at 70% (target 15%)
10/29/24-11:44:52 - Battery at 69% (target 15%)
10/29/24-11:45:52 - Battery at 69% (target 15%)
10/29/24-11:46:52 - Battery at 68% (target 15%)
10/29/24-11:47:52 - Battery at 68% (target 15%)
10/29/24-11:48:52 - Battery at 67% (target 15%)
10/29/24-11:49:52 - Battery at 66% (target 15%)
10/29/24-11:50:52 - Battery at 66% (target 15%)
10/29/24-11:51:53 - Battery at 65% (target 15%)
10/29/24-11:52:53 - Battery at 65% (target 15%)
10/29/24-11:53:53 - Battery at 64% (target 15%)
10/29/24-11:54:53 - Battery at 63% (target 15%)
10/29/24-11:55:53 - Battery at 63% (target 15%)
10/29/24-11:56:53 - Battery at 62% (target 15%)
10/29/24-11:57:53 - Battery at 62% (target 15%)
10/29/24-11:58:53 - Battery at 61% (target 15%)
10/29/24-11:59:53 - Battery at 60% (target 15%)
10/29/24-12:00:53 - Battery at 60% (target 15%)
10/29/24-12:01:53 - Battery at 59% (target 15%)
10/29/24-12:02:53 - Battery at 59% (target 15%)
10/29/24-12:03:53 - Battery at 58% (target 15%)
10/29/24-12:04:53 - Battery at 57% (target 15%)
10/29/24-12:05:53 - Battery at 57% (target 15%)
10/29/24-12:06:53 - Battery at 56% (target 15%)
10/29/24-12:07:53 - Battery at 56% (target 15%)
10/29/24-12:08:53 - Battery at 55% (target 15%)
10/29/24-12:09:53 - Battery at 54% (target 15%)
10/29/24-12:10:53 - Battery at 54% (target 15%)
10/29/24-12:11:53 - Battery at 53% (target 15%)
10/29/24-12:12:53 - Battery at 53% (target 15%)
10/29/24-12:13:53 - Battery at 52% (target 15%)
10/29/24-12:14:53 - Battery at 51% (target 15%)
10/29/24-12:15:53 - Battery at 51% (target 15%)
10/29/24-12:16:53 - Battery at 50% (target 15%)
10/29/24-12:17:54 - Battery at 50% (target 15%)
10/29/24-12:18:54 - Battery at 49% (target 15%)
10/29/24-12:19:54 - Battery at 48% (target 15%)
10/29/24-12:20:54 - Battery at 48% (target 15%)
10/29/24-12:21:54 - Battery at 47% (target 15%)
10/29/24-12:22:54 - Battery at 46% (target 15%)
10/29/24-12:23:54 - Battery at 46% (target 15%)
10/29/24-12:24:54 - Battery at 45% (target 15%)
10/29/24-12:25:54 - Battery at 45% (target 15%)
10/29/24-12:26:54 - Battery at 44% (target 15%)
10/29/24-12:27:54 - Battery at 43% (target 15%)
10/29/24-12:28:54 - Battery at 43% (target 15%)
10/29/24-12:29:54 - Battery at 42% (target 15%)
10/29/24-12:30:54 - Battery at 42% (target 15%)
10/29/24-12:31:54 - Battery at 41% (target 15%)
10/29/24-12:32:54 - Battery at 40% (target 15%)
10/29/24-12:33:54 - Battery at 40% (target 15%)
10/29/24-12:34:54 - Battery at 39% (target 15%)
10/29/24-12:35:54 - Battery at 39% (target 15%)
10/29/24-12:36:54 - Battery at 38% (target 15%)
10/29/24-12:37:55 - Battery at 37% (target 15%)
10/29/24-12:38:55 - Battery at 37% (target 15%)
10/29/24-12:39:55 - Battery at 36% (target 15%)
10/29/24-12:40:55 - Battery at 35% (target 15%)
10/29/24-12:41:55 - Battery at 35% (target 15%)
10/29/24-12:42:55 - Battery at 34% (target 15%)
10/29/24-12:43:55 - Battery at 34% (target 15%)
10/29/24-12:44:55 - Battery at 33% (target 15%)
10/29/24-12:45:55 - Battery at 32% (target 15%)
10/29/24-12:46:55 - Battery at 32% (target 15%)
10/29/24-12:47:55 - Battery at 31% (target 15%)
10/29/24-12:48:55 - Battery at 30% (target 15%)
10/29/24-12:49:55 - Battery at 30% (target 15%)
10/29/24-12:50:55 - Battery at 29% (target 15%)
10/29/24-12:51:55 - Battery at 29% (target 15%)
10/29/24-12:52:55 - Battery at 28% (target 15%)
10/29/24-12:53:55 - Battery at 27% (target 15%)
10/29/24-12:54:55 - Battery at 27% (target 15%)
10/29/24-12:55:55 - Battery at 26% (target 15%)
10/29/24-12:56:56 - Battery at 25% (target 15%)
10/29/24-12:57:56 - Battery at 25% (target 15%)
10/29/24-12:58:56 - Battery at 24% (target 15%)
10/29/24-12:59:56 - Battery at 24% (target 15%)
10/29/24-13:00:56 - Battery at 23% (target 15%)
10/29/24-13:01:56 - Battery at 22% (target 15%)
10/29/24-13:02:56 - Battery at 22% (target 15%)
10/29/24-13:03:56 - Battery at 21% (target 15%)
10/29/24-13:04:56 - Battery at 21% (target 15%)
10/29/24-13:05:56 - Battery at 20% (target 15%)
10/29/24-13:06:56 - Battery at 20% (target 15%)
10/29/24-13:07:56 - Battery at 19% (target 15%)
10/29/24-13:08:56 - Battery at 18% (target 15%)
10/29/24-13:09:56 - Battery at 18% (target 15%)
10/29/24-13:10:56 - Battery at 17% (target 15%)
10/29/24-13:11:56 - Battery at 16% (target 15%)
10/29/24-13:12:56 - Battery at 16% (target 15%)
10/29/24-13:13:56 - 🔼🪫 Disabling battery discharging
10/29/24-13:13:57 - MagSafe LED function invoked
10/29/24-13:13:57 - 💡 Setting magsafe color to green
10/29/24-13:13:57 - setting LED to green
Error: SMCWriteKey() = e00002bc
10/29/24-13:13:57 - Discharging completed at 15%
10/29/24-13:13:58 - Calibration: Calibration has discharged to 15%. Start charging to 100%
10/29/24-13:13:58 - 🔌🔋 Enabling battery charging
10/29/24-13:13:58 - 🔼🪫 Disabling battery discharging
10/29/24-13:14:00 - Battery at 15% (target 100%)
10/29/24-13:19:00 - Battery at 22% (target 100%)
10/29/24-13:24:00 - Battery at 29% (target 100%)
10/29/24-13:29:00 - Battery at 36% (target 100%)
10/29/24-13:34:00 - Battery at 43% (target 100%)
10/29/24-13:39:00 - Battery at 49% (target 100%)
10/29/24-13:44:00 - Battery at 54% (target 100%)
10/29/24-13:49:00 - Battery at 59% (target 100%)
10/29/24-13:54:00 - Battery at 64% (target 100%)
10/29/24-13:59:00 - Battery at 68% (target 100%)
10/29/24-14:04:00 - Battery at 72% (target 100%)
10/29/24-14:09:01 - Battery at 76% (target 100%)
10/29/24-14:14:01 - Battery at 80% (target 100%)
10/29/24-14:19:01 - Battery at 83% (target 100%)
10/29/24-14:24:01 - Battery at 86% (target 100%)
10/29/24-14:29:01 - Battery at 88% (target 100%)
10/29/24-14:34:01 - Battery at 90% (target 100%)
10/29/24-14:39:01 - Battery at 91% (target 100%)
10/29/24-14:44:01 - Battery at 93% (target 100%)
10/29/24-14:49:01 - Battery at 94% (target 100%)
10/29/24-14:54:01 - Battery at 95% (target 100%)
10/29/24-14:59:01 - Battery at 96% (target 100%)
10/29/24-15:04:01 - Battery at 97% (target 100%)
10/29/24-15:09:01 - Battery at 97% (target 100%)
10/29/24-15:14:01 - Battery at 98% (target 100%)
10/29/24-15:19:01 - Battery at 99% (target 100%)
10/29/24-15:24:01 - Battery at 100% (target 100%)
10/29/24-15:24:02 - Calibration: Calibration has charged to 100%. Waiting for one hour
10/29/24-16:24:02 - Calibration: Battery has been maintained at 100% for one hour
10/29/24-16:24:02 - Calibration: Start discharging to maintain percentage
10/29/24-16:24:02 - Error: macbook lid must be open before discharge
10/29/24-16:24:02 - Calibration Error: Discharge to 75% fail
10/29/24-16:24:03 - Recover in 5 secs, wait ...10/29/24-16:24:05 - Battery maintain is recovered`

然后我今天有更新，執行 battery update 得到以下
看起來應該是還可以再優化一下
`curl: (56) Failure writing output to destination, passed 277 returned 0
No need to update, offline version number v2.0.1 matches remote version number`

最後一個問題，我看 github issue 範本有說提供 log
但是我今天校正完想查 log，卻沒有這些文件
不曉得是不是因為我有反安裝再安裝的關係？
`tail -n 1000 ~/.battery/*/log
zsh: no matches found: /Users/lawliet/.battery/*/log`

js4jiang5, 2024-10-29:
請問你的機型跟年份是什麼？會出現 SMCWriteKey 表示這個想要控制的 key 在你的機型是沒有的。我自己有跑過 M1 跟 Intel都沒問題，照理說已經涵蓋所有了。
另外會出現那個 error，表示上蓋已經被闔上了，所以不允許放電。
log照理說都有的，就是放在 /Users/lawliet/.battery/，不知道什麼原因你會沒寫進去
battery update 只需要執行一次，他會檢查有沒有需要更新，如果你已經更新過一次，而且沒有更新版，就會出現那個訊息，說不需要更新。

Lawlietr, 2024-10-29:
我的是 2020 13" macbook pro m1
 
我有先看過 ptt 的文章
所以我放電的過程沒有外接螢幕，也是先關掉 macos 內建的電池最佳化
放電時是掛著遊戲死亡擱淺讓電量降到 15%
中間也沒有進入休眠狀態，並維持螢幕開合的狀態

我在安裝 Battery 之前就已經裝過 brew, xcode 我不曉得這有沒有關係
順便提個建議可以直接透過 brew 安裝套件
有需要才透過腳本安裝依賴

另外我查了一下，我前面是眼花沒看清楚
其實是有寫 log

```
tail -n 1000 ~/.battery/*.log
==> /Users/lawliet/.battery/battery.log <==
Error: SMCWriteKey() = e00002bc
10/29/24-16:26:33 - Starting battery maintenance at 80% with sailing to 75% 
10/29/24-16:26:33 - Not triggering discharge as it is not requested
10/29/24-16:26:33 - Charging to and maintaining at 80% from 100%
10/29/24-16:26:33 - Battery is not charging
10/29/24-16:26:33 - MagSafe LED function invoked
10/29/24-16:26:33 - 💡 Setting magsafe color to green
10/29/24-16:26:33 - setting LED to green
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
10/29/24-16:27:13 - Starting battery maintenance at 70% with sailing to 50% 
10/29/24-16:27:13 - Not triggering discharge as it is not requested
10/29/24-16:27:13 - Charging to and maintaining at 70% from 100%
10/29/24-16:27:13 - Battery is not charging
10/29/24-16:27:13 - MagSafe LED function invoked
10/29/24-16:27:13 - 💡 Setting magsafe color to green
10/29/24-16:27:13 - setting LED to green
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
10/29/24-16:52:31 - Starting battery maintenance at 77% with sailing to 50% 
10/29/24-16:52:31 - Not triggering discharge as it is not requested
10/29/24-16:52:31 - Charging to and maintaining at 77% from 100%
10/29/24-16:52:31 - Battery is not charging
10/29/24-16:52:31 - MagSafe LED function invoked
10/29/24-16:52:31 - 💡 Setting magsafe color to green
10/29/24-16:52:31 - setting LED to green
Error: SMCWriteKey() = e00002bc

==> /Users/lawliet/.battery/daily.log <==
Time      ,  Capacity,   Voltage,  Temperature,    Health,     Cycle

==> /Users/lawliet/.battery/sleepwatcher.log <==
2024/10/29-16:26:33 shutdown
```

我不知道有沒有問題
因為我下午電充到 100% 後
狀態列電池 icon 就一直顯示充電的圖
但我看設定裡的電池狀態應該是沒有在充電

```
10/29/24-19:16:33 - Battery at 100.0%, 12.582V, 35.2°C, SMC charging disabled
10/29/24-19:16:33 - Battery health 90.6%, Cycle 63
10/29/24-19:16:33 - Your battery is currently being maintained at 77% with sailing to 50%
10/29/24-19:16:33 -
```

觀察 battery status 電壓有微幅下降
但感覺有些怪怪的，電腦一直在用
但維持 100% 已經兩三小時了

js4jiang5, 2024-10-29:
當充電到 100%時，100%比較特別，會顯示在充電，即使用 AlDente 也一樣。你可以強制下 battery discharge 77，就會停止充電。其實只要不是 100 都會停止充電。
另外你的是 M1，會出現無法設定或讀取SMC還蠻奇怪的。你的 log 顯示一開始你沒開上蓋，後來開蓋後開始校正，表示上蓋的偵測是正常的，為什麼後來要從 100% 放電時卻偵測到上蓋沒開，這有點奇怪。請你幫我下幾個指令
smc -k CH0B -r
smc -k CH0C -r
smc -k BCLM -r
smc -k ACLC -r

js4jiang5, 2024-10-29:
因為放電的時候蓋住螢幕會進入睡眠啊。
看來你的Megasafe LED 燈顏色無法控制，那還好，不影響充放電功能。
你後來開蓋後下 discharge，有成功放電嗎？

js4jiang5, 2024-10-29:
Apple 顯示的電池量跟健康度都是不準確的，我所提供的值跟 CoconutBattery 的差不多，才是準確的。
螢幕蓋上就睡眠，是Apple希望使用者的使用方式，有些人用第三方軟體或是自己設定改成蓋上螢幕也不睡，但會這樣用的人畢竟是少數。我在BatteryOptimizer設計成放電前檢查一定要開蓋，是為了一般人的使用體驗，不要放電了不小心進入睡眠。
總之就是按照提示走，要校正或放電前，螢幕要打開，就不會有問題了。

Lawlietr, 2024-10-30:
那我了解了，感謝解說
今天看起來電池狀態是正常的
不過我電腦用四年健康度還有 90% 感覺好像也太高了

我不知道能不能提個建議
我是蠻希望可以增加一個命令是不判斷螢幕上蓋是否合起來的功能

第二個建議是可以放個抖內的按鈕
我覺得你的專案的確很有幫助
謝謝

closed 2024-12-07 by js4jiang5