js4jiang5/BatteryOptimizer_for_Mac #6: battery calibrate 無法放電

rory1205, 2024-10-31:
您好，在下完battery calibrate指令後，觀察了一段時間，電池並沒有放電情況，提供紀錄煩請協助看看問題

battery calibrate
10/31/24-20:46:30 - Calibration: Calibration has started! Start discharging to 15%
10/31/24-20:46:31 - Suspend in 5 secs, wait ..10/31/24-20:46:32 - Battery maintain is suspended
10/31/24-20:46:32 - Discharging to 15% from 74%
10/31/24-20:46:32 - 🔽🪫 Enabling battery discharging
10/31/24-20:46:33 - Battery at 74% (target 15%)
10/31/24-20:47:33 - Battery at 74% (target 15%)
10/31/24-20:48:33 - Battery at 74% (target 15%)
10/31/24-20:49:33 - Battery at 74% (target 15%)
10/31/24-20:50:33 - Battery at 74% (target 15%)
10/31/24-20:51:33 - Battery at 74% (target 15%)
10/31/24-20:52:34 - Battery at 74% (target 15%)
10/31/24-20:53:34 - Battery at 75% (target 15%)
10/31/24-20:54:34 - Battery at 75% (target 15%)
10/31/24-20:55:34 - Battery at 75% (target 15%)
10/31/24-20:56:34 - Battery at 75% (target 15%)
10/31/24-20:57:34 - Battery at 74% (target 15%)
10/31/24-20:58:34 - Battery at 74% (target 15%)
10/31/24-20:59:34 - Battery at 74% (target 15%)
10/31/24-21:00:34 - Battery at 74% (target 15%)
10/31/24-21:01:34 - Battery at 75% (target 15%)
10/31/24-21:02:34 - Battery at 75% (target 15%)
10/31/24-21:03:34 - Battery at 75% (target 15%)
10/31/24-21:04:34 - Battery at 75% (target 15%)
10/31/24-21:05:34 - Battery at 75% (target 15%)
10/31/24-21:06:34 - Battery at 75% (target 15%)
10/31/24-21:07:34 - Battery at 75% (target 15%)
10/31/24-21:08:34 - Battery at 75% (target 15%)
10/31/24-21:09:34 - Battery at 75% (target 15%)
10/31/24-21:10:34 - Battery at 75% (target 15%)
10/31/24-21:11:35 - Battery at 75% (target 15%)
10/31/24-21:12:35 - Battery at 75% (target 15%)
10/31/24-21:13:35 - Battery at 75% (target 15%)
10/31/24-21:14:35 - Battery at 75% (target 15%)
10/31/24-21:15:35 - Battery at 75% (target 15%)
10/31/24-21:16:35 - Battery at 75% (target 15%)

js4jiang5, 2024-10-31:
看來的確沒有放電，請幫我在 Terminal 執行幾個指令看一下結果
smc -k BCLM -r
smc -k CHWA -r
smc -k CH0B -r
smc -k CH0I -r
smc -k ACEN -r
smc -k ACLC -r

rory1205, 2024-11-01:
早上有升級新的版本，提供資訊如瓜，再麻煩您看看
battery version
v2.0.4
smc -k BCLM -r
  BCLM  [ui8 ]  100 (bytes 64)
smc -k CHWA -r
  CHWA  [    ]  no data
smc -k CH0B -r
  CH0B  [hex_]  (bytes 00)
smc -k CH0I -r
  CH0I  [    ]  no data
smc -k ACEN -r
  ACEN  [ui8 ]  1 (bytes 01)
smc -k ACLC -r
  ACLC  [    ]  no data

js4jiang5, 2024-11-01:
這看起來是充電狀態時記錄的值，因為你的問題是無法放電，可以在放電時讀取這些值嗎？

rory1205, 2024-11-01:
下指令開始放電的值如下：
smc -k BCLM -r
  BCLM  [ui8 ]  10 (bytes 0a)
smc -k CHWA -r
  CHWA  [    ]  no data
smc -k CH0B -r
  CH0B  [hex_]  (bytes 00)
smc -k CH0I -r
  CH0I  [    ]  no data
smc -k ACEN -r
  ACEN  [ui8 ]  1 (bytes 01)
smc -k ACLC -r
  ACLC  [    ]  no data

js4jiang5, 2024-11-01:
麻煩你分別在充電跟放電時幫我執行
smc -k CH0K -r
smc -k CH0C -r

rory1205, 2024-11-01:
看起來都一樣

充電時：
smc -k CH0K -r
  CH0K  [flag]  (bytes 00)
smc -k CH0C -r
  CH0C  [    ]  no data
  
放電時：
smc -k CH0K -r
  CH0K  [flag]  (bytes 00)
smc -k CH0C -r
  CH0C  [    ]  no data

js4jiang5, 2024-11-01:
謝謝，我大概知道怎麼做了，我改完之後再出 v2.0.5 給你們試試看

js4jiang5, 2024-11-01:
我已上傳 v2.0.5，請更新再試試看能不能放電

kirby925, 2024-11-01:
我的更新後出現Error: SMCWriteKey() = e00002bc

js4jiang5, 2024-11-01:
那你幫我執行 sudo smc -k CH0K -w 00 看看

js4jiang5, 2024-11-01:
另外， ptt 上跟你有同樣問題的人，說他用 v2.0.5 可以放電了。至少解決了一部分人的問題。

kirby925, 2024-11-01:
> 那你幫我執行 sudo smc -k CH0K -w 00 看看

Error: SMCWriteKey() = e00002bc

js4jiang5, 2024-11-01:
那請幫我查一下 
smc -k CH0J -r 
smc -k CH0R -r

kirby925, 2024-11-01:
> 那請幫我查一下 smc -k CH0J -r smc -k CH0R -r

 CH0J  [    ]  no data
 CH0R  [    ]  no data

js4jiang5, 2024-11-01:
你目前的作業系統是哪一版？

kirby925, 2024-11-01:
> 你目前的作業系統是哪一版？

13.6.9（22G830）

js4jiang5, 2024-11-01:
13.6.9 是 Ventura，我想這跟作業系統可能也有些關係，ptt上說可以的那位是 Sequoia，機型是 2020，只是為了這個升級作業系統好像也不太值得。不過我目前沒有別的辦法，能夠控制充電至少能做 sail mode 限制充電，只是無法做自動校正。

kirby925, 2024-11-01:
我是2017的ＭＢＰ 也升不上去了ＸＤ

js4jiang5, 2024-11-01:
那你更新到 v2.0.5 版可以放電嗎？

> 我是2017的ＭＢＰ 也升不上去了ＸＤ

kirby925, 2024-11-01:
我已經是2.0.5的版本了

js4jiang5, 2024-11-01:
還有一點你們可以試試看，安裝 AlDente 免費版，免費版的功能很陽春，但是可以手動放電。你們做一次手動放電，然後關掉 AlDente，再試一次我的 v2.0.5版，看是不是就能放電了。因為印象中我的 MBPR 2014 一開始也是不行，後來可以，或許跟中途安裝了AlDente 有關，他們可能可以動到到我目前不知道的內部設定。只要能放電。就能用v2.0.5做自動校正了。

js4jiang5, 2024-11-01:
對了，用上述方法，你們要記得把蘋果內建的電池最佳化關掉。

kirby925, 2024-11-01:
用你的app要關最佳化嗎？
我目前是關閉的

js4jiang5, 2024-11-01:
關閉比較好

justram, 2024-11-01:
回報一下, PTT的Sequoia owner

## Env
Model: 2019 late Intel 
OS: Sequoia 15.0.1

## Context
原本用v2.0.5可以從90%成功放電, 但是會卡在50%
用`smc -k ACEN -r`看發現 AC charge狀態是 enabled `ACEN  [ui8 ]  1 (bytes 01)`
試著用`sudo smc -k ACEN -w 00`還是不能改成disabled
按照建議裝AlDente Free再用一次`smc -k ACEN -r`看就會是 `ACEN  [ui8 ]  0 (bytes 00)`
可以繼續放電

rory1205, 2024-11-01:
同上，原本順順的放，到50%以後又慢慢的充回去，原本想多等一下提供數據的，看樓上有貼了，就上來+1一下，
我也試試用AIDente 放看看

js4jiang5, 2024-11-01:
看來Mac 不同版本差異很大，我的就算是低於50%還是繼續放。明天我再出一版 v2.0.6，你們再試試看。
AlDente 在放電這個領域的確有過人之處，看論壇中有人討論，都說可惜他們沒有 open source，不知道怎麼做的。

js4jiang5, 2024-11-01:
@rory1205 ，我把你跟 @kirby925 的回答搞在一起了，請幫我查一下
smc -k CH0J -r
smc -k CH0R -r

> 同上，原本順順的放，到50%以後又慢慢的充回去，原本想多等一下提供數據的，看樓上有貼了，就上來+1一下， 我也試試用AIDente 放看看
> 
> 回報一下， 在用您的程式放電時，我的線是顯示3W 使用AIDente 放電時，顯示的是0W 不知道這資訊對您有沒有幫助。

js4jiang5, 2024-11-01:
我大概猜到原因了，待會改一版 v2.0.6，改好再通知你們

js4jiang5, 2024-11-02:
v2.0.6 已經上傳了，請更新再看看低於 50% 能不能繼續放電

b6mh3, 2024-11-02:
PTT的b6mh19841227
MacOS: 12.7.5
BatteryOptimizer_for_MAC: 2.0.6
issue: 一樣無法放電
battery status:
11/02/24-14:31:24 - Battery at 69.8%, 11.532V, 43.3°C, no charging
11/02/24-14:31:24 - Battery health 104.4%, Cycle 9
11/02/24-14:31:24 - Your battery is currently being maintained at 80% with sailing to 20%
11/02/24-14:31:24 - You haven't scheduled calibration yet
---按照前一樓的指令---
smc -k CH0J -r:
  CH0J  [    ]  no data
smc -k CH0R -r:
  CH0R  [    ]  no data
---附註---
另外我在PTT提到的，解除安裝軟體，設定似乎還會在，所以有可能造成user忘記安裝此軟體設定過結果後續無法正常充滿

js4jiang5, 2024-11-02:
那下面兩個指令呢？
smc -k CH0K -r
smc -k CH0B -r

> PTT的b6mh19841227 MacOS: 12.7.5 BatteryOptimizer_for_MAC: 2.0.6 issue: 一樣無法放電 battery status: 11/02/24-14:31:24 - Battery at 69.8%, 11.532V, 43.3°C, no charging 11/02/24-14:31:24 - Battery health 104.4%, Cycle 9 11/02/24-

rory1205, 2024-11-02:
> smc -k CH0J -r

2.0.6 版 一樣到50% 才開始充回來

smc -k CH0J -r
  CH0J  [    ]  no data
smc -k CH0R -r
  CH0R  [    ]  no data
smc -k CH0K -r
  CH0K  [flag]  (bytes 00)
smc -k CH0B -r
  CH0B  [hex_]  (bytes 00)
smc -k CH0R -r
  CH0R  [    ]  no data

js4jiang5, 2024-11-02:
我試著寫不同的值看看，不過這次 update 請你們打指令
battery update force
這樣就不會確認版號，強制更新。因為內容有變但版號沒變，其他人就不會收到更新通知，比較不會影響其他人
現在是 v2.0.7, beta version intel_v001

> > smc -k CH0J -r
> 
> 2.0.6 版 一樣到50% 就開始充回來
> 
> smc -k CH0J -r CH0J [ ] no data smc -k CH0R -r CH0R [ ] no data smc -k CH0K -r CH0K [flag] (bytes 00) smc -k CH0B -r CH0B [hex_] (bytes 00)

justram, 2024-11-02:
用v2.0.7 intel_v001版, 一樣會卡在50%
```
❯ battery beta_version
intel_v001
❯ smc -k CH0J -r
  CH0J  [    ]  no data
❯ smc -k CH0R -r
  CH0R  [    ]  no data
❯ smc -k CH0K -r
  CH0K  [flag]  (bytes 00)
❯ smc -k CH0B -r
  CH0B  [hex_]  (bytes 00)
❯ battery status
11/02/24-18:16:15 - Battery at 49.4%, 11.478V, 35.7°C, no charging
11/02/24-18:16:15 - Battery health 78.4%, Cycle 803
11/02/24-18:16:15 - Calibration ongoing, battery maintain is suspended
11/02/24-18:16:15 - You haven't scheduled calibration yet
```

js4jiang5, 2024-11-02:
smc -k CHLS -r 有 data 嗎？

b6mh3, 2024-11-02:
smc -k CH0K -r: CH0K  [    ]  no data
smc -k CH0B -r: CH0B  [hex_]  (bytes 00)
smc -k CHLS -r: CHLS  [    ]  no data 

battery  status:
11/02/24-21:38:30 - Battery at 69.2%, 11.521V, 43.3°C, no charging
11/02/24-21:38:30 - Battery health 104.4%, Cycle 9
11/02/24-21:38:30 - Your battery is currently being maintained at 80% with sailing to 20%
11/02/24-21:38:30 - You haven't scheduled calibration yet

> 那下面兩個指令呢？ smc -k CH0K -r smc -k CH0B -r
> 
> > PTT的b6mh19841227 MacOS: 12.7.5 BatteryOptimizer_for_MAC: 2.0.6 issue: 一樣無法放電 battery status: 11/02/24-14:31:24 - Battery at 69.8%, 11.532V, 43.3°C, no charging 11/02/24-14:31:24 - Battery health 104.4%, Cycle 9 11/02/24-

js4jiang5, 2024-11-02:
諸位，看來 intel 某些機種只能放電到 50%的問題，我目前找不到解決的辦法，只能看未來有沒有機緣了，很抱歉。

js4jiang5, 2024-11-04:
請問你們執行 smc -k BSAC -r 結果有值嗎？

b6mh3, 2024-11-04:
> 請問你們執行 smc -k BSAC -r 結果有值嗎？

BSAC  [hex_]  (bytes 33)

battery status:
11/04/24-11:45:20 - Battery at 79.8%, 11.887V, 39.3°C, no charging
11/04/24-11:45:20 - Battery health 104.4%, Cycle 9
11/04/24-11:45:20 - Your battery is currently being maintained at 80% with sailing to 50%
11/04/24-11:45:20 - You haven't scheduled calibration yet

js4jiang5, 2024-11-05:
為了避免影響到主線，我開了一個 intel branch，專門處理 Intel 放電問題。要更新到 intel branch，請依照下面兩個步驟執行
1. battery update force 更新到主線最新版
2. battery update beta intel 更新到 intel branch 最新版
完成後請執行 battery version，如果出現 v0.0.1 表示成功了
然後請試試看這個版本的放電狀況如何，目前還在做實驗，所以先不要有太高的期望
還有 Intel branch 不支援中文，以找出解法為主要目的，先不花時間在語言上。

kirby925, 2024-11-05:
> 為了避免影響到主線，我開了一個 intel branch，專門處理 Intel 放電問題。要更新到 intel branch，請依照下面兩個步驟執行
> 
> 1. battery update force 更新到主線最新版
> 2. battery update beta intel 更新到 intel branch 最新版
>    完成後請執行 battery version，如果出現 v0.0.1 表示成功了
>    然後請試試看這個版本的放電狀況如何，目前還在做實驗，所以先不要有太高的期望
>    還有 Intel branch 不支援中文，以找出解法為主要目的，先不花時間在語言上。

更新到intel 版 還是無法放電

js4jiang5, 2024-11-05:
請再執行一次  battery update beta intel
確認版本是 v0.0.2 後再試一次看看

js4jiang5, 2024-11-05:
> 無法正常放電
謝謝，請幫我執行 smc -l > smc_list
然後回覆時 attach 這個 smc_list 檔案

b6mh3, 2024-11-06:
FYI

[image]

[smc_list.zip](https://github.com/user-attachments/files/17640094/smc_list.zip)

> > 無法正常放電
> 
> 謝謝，請幫我執行 smc -l > smc_list 然後回覆時 attach 這個 smc_list 檔案

rory1205, 2024-11-06:
> 謝謝，請幫我執行 smc -l > smc_list
> 然後回覆時 attach 這個 smc_list 檔案

[image]
macOS Sequoia 15.1

[smc_list.zip](https://github.com/user-attachments/files/17640390/smc_list.zip)

kirby925, 2024-11-06:
> smc -l > smc_list

[image]

[smc_list.zip](https://github.com/user-attachments/files/17640570/smc_list.zip)

kirby925, 2024-11-06:
> > 無法正常放電
> 
> 謝謝，請幫我執行 smc -l > smc_list 然後回覆時 attach 這個 smc_list 檔案
[image]
[smc_list.zip](https://github.com/user-attachments/files/17640583/smc_list.zip)

kirby925, 2024-11-06:
> > 無法正常放電
> 
> 謝謝，請幫我執行 smc -l > smc_list 然後回覆時 attach 這個 smc_list 檔案
[image]

[smc_list.zip](https://github.com/user-attachments/files/17640589/smc_list.zip)

js4jiang5, 2024-11-06:
謝謝。沒想到 2016/2017/2019 這三年的機型差異就很大了，看來 Apple 不斷的在調整控制方式。
我初步看了一下，有一個共同點是你們的 mac 似乎處於同時充電又同時放電的詭異狀態，不知道是不是受到這個程式的影響，所以想請你們做一件事，先執行 battery maintain stop，停掉這個程式，然後重新開機，再做一次 smc -l > smc_list 後上傳。
順便問個問題，你們在執行的時候，battery maintain 有在運行嗎? 如果有的話，你們的 maintain 上限跟下限是設多少？

kirby925, 2024-11-06:
> 謝謝。沒想到 2016/2017/2019 這三年的機型差異就很大了，看來 Apple 不斷的在調整控制方式。 我初步看了一下，有一個共同點是你們的 mac 似乎處於同時充電又同時放電的詭異狀態，不知道是不是受到這個程式的影響，所以想請你們做一件事，先執行 battery maintain stop，停掉這個程式，然後重新開機，再做一次 smc -l > smc_list 後上傳。 順便問個問題，你們在執行的時候，battery maintain 有在運行嗎? 如果有的話，你們的 maintain 上限跟下限是設多少？
> 
> 補充一點，重開機前作一下 SMC reset。方法是 關機後 Shift+Control+Option+電源鍵，保持按鍵約10秒，然後釋放

SMC reset後的smc list 沒開maintain

[smc_list.zip](https://github.com/user-attachments/files/17641844/smc_list.zip)

rory1205, 2024-11-06:
battery status
11/06/24-19:29:24 - Battery at 74.7%, 11.909V, 30.7°C, discharging
11/06/24-19:29:24 - Battery health 86.2%, Cycle 256
11/06/24-19:29:24 - Battery maintain is not running
11/06/24-19:29:24 - You haven't scheduled calibration yet

停用 maintain ，並重開重置SMC
[smc_list.zip](https://github.com/user-attachments/files/17646422/smc_list.zip)

rory1205, 2024-11-06:
battery update beta intel
2024-11-06 21:23:37.724 osascript[2139:55971] +[IMKClient subclass]: chose IMKClient_Modern
2024-11-06 21:23:37.724 osascript[2139:55971] +[IMKInputSession subclass]: chose IMKInputSession_Modern
2024-11-06 21:23:45.054 osascript[2140:56182] +[IMKClient subclass]: chose IMKClient_Modern
2024-11-06 21:23:45.054 osascript[2140:56182] +[IMKInputSession subclass]: chose IMKInputSession_Modern
🔋 Starting battery update

[ 1 ] Downloading latest battery version
[ 2 ] Writing script to /usr/local/bin/battery
[ Final ] Removed temporary folder

🎉 Battery tool updated.

2024-11-06 21:23:54.959 osascript[2389:56815] +[IMKClient subclass]: chose IMKClient_Modern
2024-11-06 21:23:54.959 osascript[2389:56815] +[IMKInputSession subclass]: chose IMKInputSession_Modern
Start testing intel discharge.

[smc_result.zip](https://github.com/user-attachments/files/17647729/smc_result.zip)

b6mh3, 2024-11-07:
請問現在是直接更新到0.03嗎？我直接更新沒重開沒重置

% battery update beta intel
🔋 Starting battery update

[ 1 ] Downloading latest battery version
[ 2 ] Writing script to /usr/local/bin/battery
[ Final ] Removed temporary folder

🎉 Battery tool updated.

[smc_result.zip](https://github.com/user-attachments/files/17654395/smc_result.zip)

js4jiang5, 2024-11-07:
請執行 battery update beta intel 更新至 v0.0.4
因為牽涉到 SMC，小心起見請在執行前關閉其他應用程式，
跟上次一樣更新時會直接執行測試，過程會要求輸入一次密碼，測試時間會花好幾分鐘，完成後請將 smc_result 上傳

js4jiang5, 2024-11-08:
請更新
確認更新版本為 v0.0.5 後
測試看看能否放電至 50% 以下

js4jiang5, 2024-11-11:
還沒人測試過嗎？我覺得這次成功的機率很高。

kirby925, 2024-11-11:
我更新了 還是無法放電耶

js4jiang5, 2024-11-11:
> 我更新了 還是無法放電耶

有確認版本是 v0.0.5嗎？
有的話請在放電時執行 
smc -k ACML -r
smc -k BCML -r
smc -k ACIC -r
smc -k B0AC -r
幫我看一下結果

kirby925, 2024-11-11:
> smc -k B0AC -r
[image]

[181 more replies]