js4jiang5/BatteryOptimizer_for_Mac #8: While battery calibration, enter `battery status` in another terminal result in illegal time format

RaymondJSu, 2024-11-02:
I am not sure anyone would do this, but I found out that after discharging the battery to 15% and re-enabling charging, running the `battery status` command in another terminal results in an illegal time format error.

[image]

js4jiang5, 2024-11-02:
Have you update to v2.0.7?
I fixed a date interpretation bug in v2.0.6

js4jiang5, 2024-11-02:
Another thing that I noticed is that your battery dropped down to 13% after charging starts from 15%. That behavior is not what I expected.

RaymondJSu, 2024-11-02:
> I fixed a date interpretation bug in v2.0.6
will try again on my next battery calibration

> battery dropped down to 13% after charging starts from 15%.
I believe the reason is that I continued using the laptop while not connected to the power. when it's 13% then I reconnected the power.

closed 2024-11-07 by js4jiang5