js4jiang5/BatteryOptimizer_for_Mac #12: Password: 字串洗版

StanleyQT, 2024-11-13:
作者您好：

我安裝了此程式後，首先就先做 battery calibrate ，過了一段時間後因為筆電有閒置所以有自動關閉顯示器，回來後喚醒顯示器並輸入密碼後，就看到 terminal 裡有幾行是要求要輸入密碼但最新幾行還是 Battery at xx%，然後就會開始一直洗 Password: 的字串，之所以說是字串而不是提示是因為，這時候輸入東西都會變成明碼而不是像提示要輸入密碼時打密碼的情況，按換行也是會換行然後持續跳出，不知道該如何避免這樣的情況呢，是什麼時候會要求輸入密碼呢

我的使用情境是平常上班會使用，有時候會不在位子上有設定10分鐘閒置就會關閉顯示器，下班後就會把筆電收起來帶回家，所以也沒辦法在需要輸入密碼時就輸入

不好意思說明可能有點冗長且複雜，以下附上截圖，希望能獲得解答
OS是Sequoia 15.1
[image]

謝謝

js4jiang5, 2024-11-13:
請先執行 battery update 更新到最新版

StanleyQT, 2024-11-13:
不好意思我後來因為被洗版太多所以先 uninstall 了，不過是今天早上才裝的所以應該是最新版

js4jiang5, 2024-11-13:
> 不好意思我後來因為被洗版太多所以先 uninstall 了，不過是今天早上才裝的所以應該是最新版

了解，我研究一下

StanleyQT, 2024-11-14:
顯示器是指筆電自己的螢幕，不過我先前外接螢幕使用時也有遇過相同的問題
執行 battery calibrate 之後筆電的螢幕有維持打開，也有穩定無錯誤訊息放電好一陣子，是後來離開筆電一段時間回來後才發現錯誤訊息

js4jiang5, 2024-11-14:
第一，有沒有可能你的筆電已經進入睡眠了？請檢查一下下面這個設定，避免在顯示器關閉時自動進入睡眠有沒有打勾
[image]

第二，在做 calibration 放電已經開始進行的時候，並沒有執行任何需要密碼的指令，原本的 battery maintain 也暫停運作，只會檢查狀態，也不會下達需要密碼的指令，所以會要求輸入密碼還蠻奇怪的

第三，我有嘗試在 calibrate 時讓我的筆電螢幕十分鐘後自動關閉，但是喚醒後輸入密碼並沒有碰到跟你一樣的狀況
第四，請問你有安裝 sleepwatcher 嗎？在 Terminal 下 輸入 which sleepwatcher 結果是什麼？

StanleyQT, 2024-11-15:
1. 了解，我發現這個選項真的沒有勾起來，我會再觀察看看
2. 輸入 which sleepwatcher 的結果如圖
[image]

js4jiang5, 2024-11-15:
如果打勾以後問題解決的話，請再回覆一下，謝謝

StanleyQT, 2024-11-15:
打勾以後的確是解決了，剛剛一樣有閒置一陣子但沒有再出現一樣的情況了
不好意思打擾您了，感謝您

js4jiang5, 2024-11-15:
謝謝，透過你的經驗，我下一版會改成即使沒有打勾，也不會進入睡眠，畢竟一定有很多人不會打勾。

StanleyQT, 2024-11-18:
您好，不好意思再次打擾您
前幾天更改過該處設定後，後續的電池校正程序的確是都沒有再出現前述問題
但我今天打開電腦看到電池校正程序完成的訊息出來後，放置一段時間沒有再注意terminal，剛剛再打開時發現又被 Password: 的字串洗版，即使輸入了 Ctrl+C 後還是會一直持續洗
不知道是不是有什麼地方沒注意到

js4jiang5, 2024-11-18:
請執行 battery visudo
看會出現什麼訊息

StanleyQT, 2024-11-19:
[image]
結果如附圖，再拜託您了

js4jiang5, 2024-11-19:
下面這些指令，都不應該要求你輸入密碼。如果你執行後被要求輸入密碼，請用 "cmd"+"." 中斷跳離。最後請將結果回覆給我看一下。這些指令你執行時最好用 copy paste 的方式，避免指令輸入錯誤
1. 執行 battery maintain stop
2. 執行 battery status
3. 執行 sudo smc -k CHWA -w 01
4. 執行 sudo smc -k CHWA -w 00
5. 執行 sudo smc -k ACLC -w 03
6. 執行 sudo smc -k ACLC -w 04
7. 執行 sudo smc -k ACLC -w 01
8. 執行 sudo smc -k ACLC -w 03
9. 執行 sudo smc -k CH0B -w 00
10. 執行 sudo smc -k CH0B -w 02
11. 執行 sudo smc -k CH0C -w 00
12. 執行 sudo smc -k CH0C -w 02
13. 執行 sudo smc -k CH0I -w 01
14. 執行 sudo smc -k CH0I -w 00
15. 執行 battery maintain recover
16. 執行 battery status

最後請執行 smc -l > smc_list
然後將 user 目錄底下的 smc_list 檔案在回覆時附上

StanleyQT, 2024-11-19:
[image]

第一個就會要求輸入密碼，而且我在terminal直接按"cmd"+"."會沒反應，但也有可能是我用錯方法輸入

StanleyQT, 2024-11-19:
[image]

第二個則不會

js4jiang5, 2024-11-19:
> 第二個則不會

那請從第三個開始輸入

StanleyQT, 2024-11-19:
3. 需要密碼，要輸入密碼執行嗎

js4jiang5, 2024-11-19:
不用，我想一下

js4jiang5, 2024-11-19:
請執行 sudo cat /private/etc/sudoers.d/battery
這需要輸入密碼

StanleyQT, 2024-11-19:
[image]

結果如附圖

js4jiang5, 2024-11-19:
請輸入 sudo smc -k CH0I -r
看是否需要密碼

StanleyQT, 2024-11-19:
不需要

js4jiang5, 2024-11-19:
那 sudo smc -k CH0I -w 00 呢

StanleyQT, 2024-11-19:
也不需要

js4jiang5, 2024-11-19:
阿不對，因為你剛剛有輸入一次密碼，有時效性，所以現在看不準
要等個十分鐘後再輸入上述指令

StanleyQT, 2024-11-19:
沒問題，稍後再向您回報

js4jiang5, 2024-11-19:
在等待的期間，請先看一下你的使用者群組，你目前所使用的帳號 stanleychang 是不是沒有管理者權限？

StanleyQT, 2024-11-19:
[image]

這樣應該是管理者是嗎，但其實我們公司有導入一套由cloud納管的帳號管理系統，不確定是不是有可能因此導致權限相關的問題

js4jiang5, 2024-11-19:
有可能，難怪你的情況比較特殊。我想一下怎麼做

StanleyQT, 2024-11-19:
不好意思造成您的麻煩，如果有什麼需要我提供的資訊，例如該管理系統是否有在Mac上新增其他user，請再告知我

StanleyQT, 2024-11-19:
有的

[image]

StanleyQT, 2024-11-19:
需要

js4jiang5, 2024-11-19:
請執行 sudo visudo
page down 找到
[image]

游標移到 
%admin          ALL = (ALL) ALL
的最右邊，然後按 a 可以開始編輯
然後按 enter 換到下一行，接著輸入
stanleychang          ALL = (ALL) ALL
完成後按 ESC 跳離編輯，接著按:
然後輸入 wq!
儲存檔案

js4jiang5, 2024-11-19:
以上步驟完成後，請重新開機，然後輸入
sudo smc -k CH0I -r
看是否還需要密碼

StanleyQT, 2024-11-20:
還是需要密碼

js4jiang5, 2024-11-20:
請執行 sudo cat /etc/sudoers | grep stanleychang 確認一下

StanleyQT, 2024-11-20:
[image]

結果如圖

js4jiang5, 2024-11-20:
如果是這樣，我覺得應該是你們公司的設定造成的限制，可能需要尋求你們公司 IT 的協助了。

StanleyQT, 2024-11-20:
沒問題，沒關係那我再找他們看看
感謝您的協助

closed 2024-12-07 by js4jiang5