js4jiang5/BatteryOptimizer_for_Mac #13: Battery calibration doesn't seem to work

iammmiru, 2024-11-15:
Hi,

I've been happily using our fork on `battery` for a week or so.
Thanks for your improvements :)

I'm opening this ticket since the calibration doesn't seem to work.
The calibration is supposed to happen at 9am, and `battery status` tells me it's on going.
at 1:33pm, my battery is at 79.5%
at 2:30pm, my battery is at 79.9% (?)
at 6:23pm, it's 78.1%

My understanding is, it should discharge the battery to 15% and charge it to 100%.

It's an M1 Macbook pro, and it's plugged in with a USB-C charger (through a dongle, the apple one).

Below is the status messages

`➜ ~ battery status

11/15/24-13:44:07 - Battery at 79.5%, 12.077V, 27.4°C, no charging
11/15/24-13:44:07 - Battery health 84.3%, Cycle 85
11/15/24-13:44:07 - Calibration ongoing, battery maintain is suspended
11/15/24-13:44:07 - Schedule calibration on day 15    at 9:00
11/15/24-13:44:08 - Next calibration date is 2024/12/15

➜ ~ battery status

11/15/24-14:33:29 - Battery at 79.9%, 12.068V, 28.7°C, no charging
11/15/24-14:33:29 - Battery health 84.5%, Cycle 85
11/15/24-14:33:29 - Calibration ongoing, battery maintain is suspended
11/15/24-14:33:30 - Schedule calibration on day 15    at 9:00
11/15/24-14:33:30 - Next calibration date is 2024/12/15

➜ ~ battery status

11/15/24-18:23:25 - Battery at 78.1%, 12.001V, 29.5°C, discharging
11/15/24-18:23:25 - Battery health 85.5%, Cycle 85
11/15/24-18:23:25 - Calibration ongoing, battery maintain is suspended
11/15/24-18:23:25 - Schedule calibration on day 15    at 9:00
11/15/24-18:23:26 - Next calibration date is 2024/12/15`

Here is the log

`➜ ~ battery logs    
👾 Battery CLI logs:

11/15/24-18:08:29 - Battery at 78% (target 15%)
11/15/24-18:09:29 - Battery at 78% (target 15%)
11/15/24-18:10:29 - Battery at 78% (target 15%)
11/15/24-18:11:29 - Battery at 78% (target 15%)
11/15/24-18:12:29 - Battery at 78% (target 15%)
11/15/24-18:13:29 - Battery at 78% (target 15%)
11/15/24-18:14:29 - Battery at 78% (target 15%)
11/15/24-18:15:29 - Battery at 78% (target 15%)
11/15/24-18:16:29 - Battery at 78% (target 15%)
11/15/24-18:17:03 - MagSafe LED function invoked
11/15/24-18:17:03 - 💡 Setting magsafe color to none
11/15/24-18:17:03 - setting LED to none
11/15/24-18:17:08 - MagSafe LED function invoked
11/15/24-18:17:08 - 💡 Setting magsafe color to green
11/15/24-18:17:08 - setting LED to green
11/15/24-18:17:30 - Battery at 78% (target 15%)
11/15/24-18:18:30 - Battery at 78% (target 15%)
11/15/24-18:18:49 - MagSafe LED function invoked
11/15/24-18:18:50 - 💡 Setting magsafe color to none
11/15/24-18:18:50 - setting LED to none
11/15/24-18:19:15 - MagSafe LED function invoked
11/15/24-18:19:15 - 💡 Setting magsafe color to green
11/15/24-18:19:15 - setting LED to green
11/15/24-18:19:21 - MagSafe LED function invoked
11/15/24-18:19:21 - 💡 Setting magsafe color to none
11/15/24-18:19:21 - setting LED to none
11/15/24-18:19:26 - MagSafe LED function invoked
11/15/24-18:19:26 - 💡 Setting magsafe color to green
11/15/24-18:19:26 - setting LED to green
11/15/24-18:19:30 - Battery at 78% (target 15%)
11/15/24-18:19:46 - MagSafe LED function invoked
11/15/24-18:19:46 - 💡 Setting magsafe color to none
11/15/24-18:19:46 - setting LED to none
11/15/24-18:19:51 - MagSafe LED function invoked
11/15/24-18:19:51 - 💡 Setting magsafe color to green
11/15/24-18:19:51 - setting LED to green
11/15/24-18:20:30 - Battery at 78% (target 15%)
11/15/24-18:21:08 - MagSafe LED function invoked
11/15/24-18:21:08 - 💡 Setting magsafe color to none
11/15/24-18:21:08 - setting LED to none
11/15/24-18:21:31 - Battery at 78% (target 15%)
11/15/24-18:22:31 - Battery at 78% (target 15%)
11/15/24-18:23:32 - Battery at 78% (target 15%)
11/15/24-18:24:04 - MagSafe LED function invoked
11/15/24-18:24:04 - 💡 Setting magsafe color to green
11/15/24-18:24:04 - setting LED to green
11/15/24-18:24:09 - MagSafe LED function invoked
11/15/24-18:24:09 - 💡 Setting magsafe color to none
11/15/24-18:24:09 - setting LED to none
11/15/24-18:24:14 - MagSafe LED function invoked
11/15/24-18:24:14 - 💡 Setting magsafe color to green
11/15/24-18:24:14 - setting LED to green
11/15/24-18:24:19 - MagSafe LED function invoked
11/15/24-18:24:19 - 💡 Setting magsafe color to none
11/15/24-18:24:19 - setting LED to none
11/15/24-18:24:32 - Battery at 78% (target 15%)
11/15/24-18:24:35 - MagSafe LED function invoked
11/15/24-18:24:35 - 💡 Setting magsafe color to green
11/15/24-18:24:35 - setting LED to green
11/15/24-18:24:40 - MagSafe LED function invoked
11/15/24-18:24:40 - 💡 Setting magsafe color to none
11/15/24-18:24:40 - setting LED to none
11/15/24-18:24:45 - MagSafe LED function invoked
11/15/24-18:24:45 - 💡 Setting magsafe color to green
11/15/24-18:24:45 - setting LED to green
11/15/24-18:24:50 - MagSafe LED function invoked
11/15/24-18:24:50 - 💡 Setting magsafe color to none
11/15/24-18:24:50 - setting LED to none
11/15/24-18:25:33 - Battery at 78% (target 15%)
11/15/24-18:25:36 - MagSafe LED function invoked
11/15/24-18:25:36 - 💡 Setting magsafe color to green
11/15/24-18:25:36 - setting LED to green
11/15/24-18:26:33 - Battery at 78% (target 15%)
11/15/24-18:27:33 - Battery at 78% (target 15%)
11/15/24-18:28:33 - Battery at 78% (target 15%)
11/15/24-18:29:33 - Battery at 78% (target 15%)
11/15/24-18:30:33 - Battery at 78% (target 15%)
11/15/24-18:31:33 - Battery at 78% (target 15%)
11/15/24-18:32:33 - Battery at 78% (target 15%)
11/15/24-18:33:33 - Battery at 78% (target 15%)
11/15/24-18:34:34 - Battery at 78% (target 15%)
11/15/24-18:35:34 - Battery at 78% (target 15%)
11/15/24-18:36:34 - Battery at 78% (target 15%)
11/15/24-18:37:34 - Battery at 78% (target 15%)
11/15/24-18:38:34 - Battery at 78% (target 15%)
11/15/24-18:39:34 - Battery at 78% (target 15%)
11/15/24-18:40:34 - Battery at 78% (target 15%)
11/15/24-18:41:34 - Battery at 78% (target 15%)
11/15/24-18:42:34 - Battery at 78% (target 15%)
11/15/24-18:43:34 - Battery at 78% (target 15%)
11/15/24-18:44:35 - Battery at 78% (target 15%)
11/15/24-18:45:35 - Battery at 78% (target 15%)
11/15/24-18:46:35 - Battery at 78% (target 15%)
11/15/24-18:47:35 - Battery at 78% (target 15%)
11/15/24-18:48:35 - Battery at 78% (target 15%)
11/15/24-18:49:35 - Battery at 78% (target 15%)
11/15/24-18:50:35 - Battery at 78% (target 15%)
11/15/24-18:51:35 - Battery at 78% (target 15%)
11/15/24-18:52:35 - Battery at 78% (target 15%)
11/15/24-18:53:36 - Battery at 78% (target 15%)

🖥️	Battery GUI logs:

tail: /Users/mirulee/.battery/gui.log: No such file or directory

📁 Config folder details:

total 344
drwxr-xr-x@ 13 mirulee  staff   416B Nov 15 10:23 .
drwxr-x---+ 66 mirulee  staff   2.1K Nov 13 19:33 ..
-rwxr-xr-x@  1 mirulee  staff    98K Nov 15 18:53 battery.log
-rwxr-xr-x@  1 mirulee  staff     6B Nov  8 19:21 battery.pid
-rw-r--r--@  1 mirulee  staff     6B Nov 15 10:23 calibrate.pid
-rw-r--r--@  1 mirulee  staff    42B Nov  8 19:22 calibrate_schedule
-rw-r--r--@  1 mirulee  staff   1.0K Nov 15 00:06 daily.log
-rw-r--r--@  1 mirulee  staff     8B Nov 11 19:36 informed.version
-rw-r--r--@  1 mirulee  staff     3B Nov  8 11:37 maintain.percentage
-rw-r--r--@  1 mirulee  staff    20B Nov 15 10:23 sig
-rw-r--r--@  1 mirulee  staff     6B Nov 15 10:23 sig.pid
-rw-r--r--@  1 mirulee  staff   1.3K Nov 15 10:23 sleepwatcher.log
-rw-r--r--@  1 mirulee  staff    10B Nov 15 10:23 state

⚙️	Battery data:

11/15/24-18:53:43 - Battery at 77.6%, 12.025V, 29.1°C, no charging
11/15/24-18:53:43 - Battery health 84.4%, Cycle 85
11/15/24-18:53:44 - Calibration ongoing, battery maintain is suspended
11/15/24-18:53:44 - Schedule calibration on day 15    at 9:00
11/15/24-18:53:44 - Next calibration date is 2024/12/15

Battery CLI utility v2.0.10`

Thanks in advance

js4jiang5, 2024-11-16:
1. could you run command "battery version" and let me know what your version is?
2. according to the log, it frequently varied between "no charging" and "discharging". One thing that I'm sure is the discharge command is conducted one time only at the beginning of discharge, so the frequent change is not caused by the tool. The truth that it sometimes changed to discharging proves that discharge command was conducted successfully.
3. if calibration is still ongoing, please run command "smc -l > smc_list" to capture the whole smc status and attach the file smc_list in your next reply.
4. after capturing smc_list successfully, please stop current calibration by command "battery calibrate stop"
5. if your version is not v2.0.11, please run command "battery update" to update to the latest version. Otherwise just run "battery maintain recover" to resume maintain
6. run "battery status to check if maintain is running"
7. right now I'm suspicious of the USB-C charging source you used. If possible, please replace it with magsafe connector power supply as power source, and run command "battery discharge 75" to check if it can successfully discharge to 75%
8. please let me know if you have disable Apple's battery optimization.

js4jiang5, 2024-11-20:
> Thanks for the response.
> 
> 1. I believe it was 2.0.10, and now I updated to 2.0.12.
> 2. I stopped the calibration, but I'll include the smc_list after I try calibration again (perhaps tomorrow).
> 3. This laptop is mostly powered by the usb-c charging. Although I'll try with megsafe, I'd really appreciate if the calibration works well with usb-c charging as well. The maintenance functionality works without any problem, btw.
> 4. yes. it's disabled.

It's possible that 2.0.12 can solve your problem.
Previous modification for intel Macs might affect the discharge function of Apple Macs.
Some other person encounter similar issue with you, and that's because he is running AlDente at the same time and cause conflict. So if you've installed AlDente, you'll need to turn it off.

iammmiru, 2024-11-21:
> It's possible that 2.0.12 can solve your problem.

It seems the calibration is working with 2.0.12.

➜ ~ battery status

11/21/24-14:42:02 - Battery at 17.3%, 11.041V, 25.9°C, discharging
11/21/24-14:42:02 - Battery health 84.1%, Cycle 86
11/21/24-14:42:02 - Calibration ongoing, battery maintain is suspended
11/21/24-14:42:02 - Your calibration schedule is disabled. Enable it by
11/21/24-14:42:02 - battery schedule enable

thanks for your help :)

closed 2024-12-07 by js4jiang5