js4jiang5/BatteryOptimizer_for_Mac #16: Daily Log 某些日期不記錄

raymao9, 2024-12-12:
**Your macbook model and year:**
Macbook M1Pro 2021 (A2442)

**Your version number: (run "battery version" to find out)**
2.0.12

**Your battery status: (run "battery status" to find out)**
最新日誌如下
```
12/13/24-01:08:56 - 電池目前 55.7%, 11.565V, 27.7°C, 放電中
12/13/24-01:08:56 - 電池健康度 99.1%, 循環次數 60
12/13/24-01:08:56 - 您的電池最佳化維持在 60% 滑行至 55%
12/13/24-01:08:56 - 電池自動校正時程安排在 20 日 10:00 開始
12/13/24-01:08:56 - 下次校正日期是 2024/12/20
```

**What is the issue?**
使用一段時間了，發現 Daily Log 在某些日期不記錄，我電腦的使用率每天平均 8 小時以上，很偶爾才會低於 5 小時，沒有整整一天都不開蓋使用的情況，那麼這些不記錄的日期是有什麼情況導致它 `漏記` 嗎？尤其像 `12/1 ~ 12/6` 整整一週都沒記，`12/8 ~ 12/10` 三天也沒記，就蠻奇怪的，我從 11/19 決定使用後也沒有再移除過此工具，那會是什麼原因？

Daily Log 輸出如下
```
Time      ,  Capacity,   Voltage,  Temperature,    Health,     Cycle
2024/11/19,     80.1%,   12.263V,       33.1°C,     99.8%,        56
2024/11/21,     64.0%,   11.879V,       25.6°C,    100.8%,        57
2024/11/22,     60.2%,   11.728V,       25.5°C,     99.9%,        57
2024/11/23,     67.5%,   11.899V,       22.9°C,     99.5%,        57
2024/11/23,     67.5%,   11.899V,       22.9°C,     99.5%,        57
2024/11/23,     64.0%,   12.267V,       27.0°C,     99.3%,        58
2024/11/24,     52.7%,    11.64V,       23.3°C,    100.6%,        58
2024/11/25,     53.6%,   11.547V,       29.1°C,     99.0%,        58
2024/11/26,     56.4%,   11.694V,       27.5°C,    101.0%,        58
2024/11/27,     56.2%,   11.639V,       25.1°C,    100.0%,        58
2024/11/29,     53.3%,   11.628V,       25.5°C,    100.3%,        58
2024/11/30,     65.0%,    11.87V,       23.6°C,    100.3%,        59
2024/12/07,     61.9%,   11.831V,       26.0°C,     99.8%,        59
2024/12/11,     52.2%,   11.533V,       25.8°C,     99.5%,        60
2024/12/12,     57.8%,   11.759V,       28.5°C,    100.4%,        60
2024/12/13,     58.0%,   11.755V,       25.6°C,     99.7%,        60
```

**Error logs: (run "battery logs" to find out)**

js4jiang5, 2024-12-12:
目前的設定是在凌晨12點時記錄，如果當時沒開機就不會記錄。
沒有考慮到沒有24小時開機的狀況，下一版會改成如果當天都還沒記錄，就會做記錄，謝謝。

raymao9, 2024-12-13:
> 目前的設定是在凌晨12點時記錄，如果當時沒開機就不會記錄。 沒有考慮到沒有24小時開機的狀況，下一版會改成如果當天都還沒記錄，就會做記錄，謝謝。 我陸續都有在修正一些問題，目前最新版是 v2.0.14，建議你收到更新通知時最好執行 battery update 更新一下。

了解，原來如此~

closed 2024-12-13 by raymao9