js4jiang5/BatteryOptimizer_for_Mac #17: 請問電池最佳化暫停後，什麼時候會開始充電

ogre0403, 2024-12-24:
**Your macbook model and year:**
Macbook Pro M3 Max

**Your version number: (run "battery version" to find out)**

Battery CLI utility v2.0.14

**Your battery status: (run "battery status" to find out)**

```
12/24/24-13:46:49 - 電池目前 76.7%, 12.404V, 25.8°C, 暫停充電
12/24/24-13:46:49 - 電池健康度 101.4%, 循環次數 6
12/24/24-13:46:49 - 電池最佳化已暫停
12/24/24-13:46:49 - 您尚未設定電池自動校正時程
```

**What is the issue?**
您好，原本在 maintain mode，想臨時充到100%，我看help 文件，可以透過suspend 暫停後進行充電，但我執行 ` battery maintain suspend` 後, 還是一直停在`暫停充電`。請問是不是有什麼條件下才會開始充電。  

[image]

另外，我也嘗試用 `battery charge 100`，但command會停在 `setting LED to orange`。且電量都沒改變。

[image]

**Error logs: (run "battery logs" to find out)**

```
12/24/24-13:27:40 - Charging to and maintaining at 80% from 77%
12/24/24-13:27:40 - MagSafe LED function invoked
12/24/24-13:27:40 - 💡 Setting magsafe color to green
12/24/24-13:27:40 - setting LED to green
12/24/24-13:27:40 - MagSafe LED function invoked
12/24/24-13:27:40 - 💡 Setting magsafe color to green
12/24/24-13:27:40 - setting LED to green
12/24/24-13:27:47 - Killing old maintain process at 34863 active
12/24/24-13:27:47 - Killing running maintain daemons & enabling charging as default state
12/24/24-13:27:47 - Disabling daemon at gui/501/com.battery.app
12/24/24-13:28:07 - New process ID 35518
12/24/24-13:28:07 - Writing new setting 80 to maintain_percentage
12/24/24-13:28:07 - Daemon already exists, checking for differences
12/24/24-13:28:07 - Starting battery maintenance at 80% with sailing to 75%
12/24/24-13:28:07 - Not triggering discharge as it is not requested
12/24/24-13:28:07 - Charging to and maintaining at 80% from 77%
12/24/24-13:28:33 - ack battery maintain suspend
12/24/24-13:29:37 - 🔼🪫 Disabling battery discharging
12/24/24-13:29:38 - ack battery maintain recover
12/24/24-13:29:38 - MagSafe LED function invoked
12/24/24-13:29:38 - 💡 Setting magsafe color to green
12/24/24-13:29:38 - setting LED to green
12/24/24-13:29:55 - Killing old maintain process at 35518 active
12/24/24-13:29:55 - New process ID 36790
12/24/24-13:29:55 - Writing new setting 100 to maintain_percentage
12/24/24-13:29:55 - Daemon already exists, checking for differences
12/24/24-13:29:55 - Starting battery maintenance at 100% with sailing to 95%
12/24/24-13:29:55 - Not triggering discharge as it is not requested
12/24/24-13:29:55 - Charging to and maintaining at 100% from 77%
12/24/24-13:29:55 - Charge below 95
12/24/24-13:29:55 - 🔼🪫 Disabling battery discharging
12/24/24-13:29:56 - 🔌🔋 Enabling battery charging
12/24/24-13:30:02 - Charge below 95
12/24/24-13:30:02 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:03 - 🔌🔋 Enabling battery charging
12/24/24-13:30:09 - Charge below 95
12/24/24-13:30:09 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:10 - 🔌🔋 Enabling battery charging
12/24/24-13:30:16 - Charge below 95
12/24/24-13:30:16 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:17 - 🔌🔋 Enabling battery charging
12/24/24-13:30:24 - Charge below 95
12/24/24-13:30:24 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:25 - 🔌🔋 Enabling battery charging
12/24/24-13:30:31 - Charge below 95
12/24/24-13:30:31 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:32 - 🔌🔋 Enabling battery charging
12/24/24-13:30:38 - Charge below 95
12/24/24-13:30:38 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:39 - 🔌🔋 Enabling battery charging
12/24/24-13:30:45 - Charge below 95
12/24/24-13:30:45 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:46 - 🔌🔋 Enabling battery charging
12/24/24-13:30:48 - Killing old maintain process at 36790 active
12/24/24-13:30:48 - New process ID 37982
12/24/24-13:30:48 - Writing new setting 90 to maintain_percentage
12/24/24-13:30:48 - Daemon already exists, checking for differences
12/24/24-13:30:48 - Starting battery maintenance at 90% with sailing to 85%
12/24/24-13:30:48 - Not triggering discharge as it is not requested
12/24/24-13:30:48 - Charging to and maintaining at 90% from 77%
12/24/24-13:30:48 - Charge below 85
12/24/24-13:30:48 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:49 - 🔌🔋 Enabling battery charging
12/24/24-13:30:55 - Charge below 85
12/24/24-13:30:55 - 🔼🪫 Disabling battery discharging
12/24/24-13:30:56 - 🔌🔋 Enabling battery charging
12/24/24-13:31:02 - Charge below 85
12/24/24-13:31:02 - 🔼🪫 Disabling battery discharging
12/24/24-13:31:03 - 🔌🔋 Enabling battery charging
12/24/24-13:31:10 - Charge below 85
12/24/24-13:31:10 - 🔼🪫 Disabling battery discharging
12/24/24-13:31:11 - 🔌🔋 Enabling battery charging
12/24/24-13:31:15 - Killing old maintain process at 37982 active
12/24/24-13:31:15 - New process ID 38571
12/24/24-13:31:15 - Writing new setting 80 to maintain_percentage
12/24/24-13:31:15 - Daemon already exists, checking for differences
12/24/24-13:31:15 - Starting battery maintenance at 80% with sailing to 75%
12/24/24-13:31:15 - Not triggering discharge as it is not requested
12/24/24-13:31:15 - Charging to and maintaining at 80% from 77%
12/24/24-13:31:44 - 🔼🪫 Disabling battery discharging
12/24/24-13:31:45 - 🔌🔋 Enabling battery charging
12/24/24-13:31:46 - ack battery maintain suspend
12/24/24-13:40:55 - Killing old maintain process at 38571 suspended
12/24/24-13:41:01 - Killing old maintain process at 38571 suspended
12/24/24-13:41:01 - New process ID 40271
12/24/24-13:41:01 - Daemon already exists, checking for differences
12/24/24-13:41:01 - Debug trail. User: ogre0403, config folder: /Users/ogre0403/.battery, logfile: /Users/ogre0403/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
12/24/24-13:41:01 - Recovering maintenance percentage 80
12/24/24-13:41:01 - Starting battery maintenance at 80% with sailing to 75%
12/24/24-13:41:01 - Not triggering discharge as it is not requested
12/24/24-13:41:01 - Charging to and maintaining at 80% from 77%
12/24/24-13:41:01 - MagSafe LED function invoked
12/24/24-13:41:01 - 💡 Setting magsafe color to green
12/24/24-13:41:01 - setting LED to green
12/24/24-13:41:01 - MagSafe LED function invoked
12/24/24-13:41:01 - 💡 Setting magsafe color to green
12/24/24-13:41:01 - setting LED to green
12/24/24-13:41:20 - 🔼🪫 Disabling battery discharging
12/24/24-13:41:21 - 🔌🔋 Enabling battery charging
12/24/24-13:41:22 - ack battery maintain suspend

🖥️	Battery GUI logs:

📁 Config folder details:

total 88
drwxr-xr-x@  8 ogre0403  staff   256B 12 24 13:31 .
drwxr-x---+ 44 ogre0403  staff   1.4K 12 24 13:44 ..
-rwxr-xr-x@  1 ogre0403  staff    20K 12 24 13:44 battery.log
-rw-r--r--@  1 ogre0403  staff    16B 12 24 13:41 battery.pid
-rw-r--r--@  1 ogre0403  staff   143B 12 24 13:31 config_battery
-rw-r--r--@  1 ogre0403  staff   629B 12 24 00:40 daily.log
-rw-r--r--@  1 ogre0403  staff    14B 12 24 13:41 sig.pid
-rw-r--r--@  1 ogre0403  staff   4.3K 12 24 13:32 sleepwatcher.log

⚙️	Battery data:

12/24/24-13:44:41 - 電池目前 76.7%, 12.4V, 25.7°C, 暫停充電
12/24/24-13:44:41 - 電池健康度 101.4%, 循環次數 6
12/24/24-13:44:41 - 電池最佳化已暫停
12/24/24-13:44:41 - 您尚未設定電池自動校正時程

Battery CLI utility v2.0.14
```

js4jiang5, 2024-12-24:
正常來說只要下達 battery maintain suspend，就會立刻開始充電
從 log 來看你的系統確實有異常，因為 "Enabling battery charging" 表示已經有下達開始充電指令，但卻沒有充電
請執行 smc -k CH0C -r 看結果如何。
另外，通常系統異常表示你的 SMC 出了問題，M3 只要重開機就會重置 SMC，大部分的問題都會獲得解決。

ogre0403, 2024-12-24:
```
 smc -k CH0C -r                                                                                                                            
 CH0C  [hex_]  (bytes 02)
```

js4jiang5, 2024-12-24:
> 重開機前的結果
> 
> ```shell
>  smc -k CH0C -r                                                                                                                            
>  CH0C  [hex_]  (bytes 02)
> ```
> 
> 重啟後的結果不一樣
> 
> ```shell
> smc -k CH0C -r                                                                                                              
>  CH0C  [hex_]  (bytes 00)
> ```

02 表示停止充電
00 表示正在充電

不過如果你要確認是否運作正常
先執行 battery maintain recover ，確認電池是暫停充電後
再執行 battery maintain suspend 看看是否有正常充電

ogre0403, 2024-12-24:
感謝，重開機後就正常了。

closed 2024-12-24 by ogre0403