js4jiang5/BatteryOptimizer_for_Mac #20: unary operator expected

twixfix, 2025-01-24:
MacBook Air M3 - battery v2.0.18

01/24/25-15:53:05 - Battery at 48.2%, 11.477V, 26.5°C, discharging
01/24/25-15:53:05 - Battery health 101.2%, Cycle 8
01/24/25-15:53:05 - Your battery is currently being maintained at 80% with sailing to 75%
01/24/25-15:53:05 - Schedule calibration on day 1 at 9:00
01/24/25-15:53:05 - Next calibration date is 2025/02/01

Tried to set schedule and got an error but calibrations was still scheduled

% battery schedule
/usr/local/bin/battery: line 2437: [: ==: unary operator expected
/usr/local/bin/battery: line 2453: [: ==: unary operator expected

01/24/25-15:51:37 - Schedule calibration on day 1 at 9:00
01/24/25-15:51:37 - Next calibration date is 2025/02/01

js4jiang5, 2025-01-29:
I'll check after lunar new year vacation.

js4jiang5, 2025-01-30:
Thank you for pointing it out. I'll fix it in v 2.0.19.

js4jiang5, 2025-01-30:
I've uploaded v2.0.19. You may run "battery update" to update to v2.0.19.

closed 2025-01-30 by twixfix