js4jiang5/BatteryOptimizer_for_Mac #21: dist/README.md out of date

chkpnt, 2025-01-24:
I've followed the instructions at https://github.com/js4jiang5/BatteryOptimizer_for_MAC/blob/main/dist/README.md, but it doesn't work:

```
$ make
make: *** No targets specified and no makefile found.  Stop.
```

So I guess the README.md is out of date. How did you built smc and smc_intel?

js4jiang5, 2025-01-29:
I'll get back to you after lunar new year vacation.

js4jiang5, 2025-01-30:
> I've followed the instructions at https://github.com/js4jiang5/BatteryOptimizer_for_MAC/blob/main/dist/README.md, but it doesn't work:
> 
> ```
> $ make
> make: *** No targets specified and no makefile found.  Stop.
> ```
> 
> So I guess the README.md is out of date. How did you built smc and smc_intel?

It's not necessary to generate smc binary by yourself since it's already included in installation.
If you're intereted in smc binaray but failed to generate it, you should consult the designer 

[smcFanControl](https://github.com/hholtmann/smcFanControl.git)

chkpnt, 2025-01-30:
> It's not necessary to generate smc binary by yourself since it's already included in installation.

Yeah, I know. All the source code I can skim for dubious things, but I can't verify if the smc binaries contains any malware. Did you build it by your own or was it prebuilt by someone else, so I can compare the hash values?  
I've looked into the latest release of [smcFanControl](https://github.com/hholtmann/smcFanControl/releases/tag/2.6), but it contains only a binary `smc` (which is much small than the `smc` included in this repo), not `smc_intel`.

Please don't get me wrong, I don't want to insinuate anything!

closed 2025-02-17 by js4jiang5