js4jiang5/BatteryOptimizer_for_Mac #22: 到了預定的校正時間沒有執行校正

raymao9, 2025-01-25:
**Your macbook model and year:**
Apple M1 A2442

**Your version number: (run "battery version" to find out)**
v2.0.18

**Your battery status: (run "battery status" to find out)**

```
01/26/25-01:00:19 - 電池目前 77.5%, 12.18V, 27.5°C, 暫停充電
01/26/25-01:00:19 - 電池健康度 98.5%, 循環次數 79
01/26/25-01:00:19 - 您的電池最佳化維持在 80% 滑行至 75%
01/26/25-01:00:19 - 電池自動校正時程安排在 25 日 10:00 開始
01/26/25-01:00:19 - 下次校正日期是 2025/02/25
```

**What is the issue?**
1 月初將電池自動校正時間改為每月 25 日，所以預定 1/25 10:00 要進行校正
如下圖所示，電腦是在 25 日的凌晨 00:01 通知的

[image]

但我是在 25 日的 10:30 左右才打開電腦，按之前的說明和理解，超過時間後，開蓋就應該立即執行
但從 `battery logs` 記錄來看，無事發生
下面貼 logs

**Error logs: (run "battery logs" to find out)**

以下是 24-25 日 一整天的記錄
```
01/24/25-09:56:41 - Charge below 75
01/24/25-09:56:41 - 🔼🪫 Disabling battery discharging
01/24/25-09:56:42 - 🔌🔋 Enabling battery charging
01/24/25-09:56:49 - 💡 Setting magsafe color to orange
01/24/25-10:02:39 - Stop charge above 80
01/24/25-10:02:39 - 🔌🪫 Disabling battery charging
01/24/25-10:03:40 - 💡 Setting magsafe color to green
01/25/25-01:52:37 - 💡 Setting magsafe color to none
01/25/25-06:00:59 - 💡 Setting magsafe color to green
01/25/25-07:38:12 - 💡 Setting magsafe color to none
01/25/25-08:39:38 - 💡 Setting magsafe color to green
01/25/25-08:40:38 - 💡 Setting magsafe color to green
01/25/25-19:16:31 - 💡 Setting magsafe color to none
01/25/25-20:08:29 - 💡 Setting magsafe color to green
```

這樣應該是沒有進行過校正吧？
然後再看今天最新記錄，下次校正日期已經自動跳到 2/25 了

```
01/26/25-01:00:19 - 電池目前 77.5%, 12.18V, 27.5°C, 暫停充電
01/26/25-01:00:19 - 電池健康度 98.5%, 循環次數 79
01/26/25-01:00:19 - 您的電池最佳化維持在 80% 滑行至 75%
01/26/25-01:00:19 - 電池自動校正時程安排在 25 日 10:00 開始
01/26/25-01:00:19 - 下次校正日期是 2025/02/25
```

另附上我在 12/27 手動執行過的校正記錄

```
Time            , Completed, Health_before, Health_after, Duration/Error
2024/12/27 20:35,       Yes,         99.8%,        99.7%, 22 小時 15 分 
```

此外
我知道這個程式平常手動執行的時候會用終端機，但自動在背景的時候會不會用到？
我這幾天大概一整天都在用終端機執行別的工作，我不知道這有沒有影響。

raymao9, 2025-01-28:
我自己推測可能是時區的問題，我 12/27 那天校正完後指定了下次校正時間是每月 25 日，但 12/27 那天人在溫哥華，而計劃應該執行校正的 1/25 人在台北，電腦時區是自動的，按理說不應該有差，但我猜軟體本身可能沒針對時區做調整，按它提示預備校正的時間都是 `00:00` 來看，這種提示時間就不太合理，但如果按台北時間，這個校正提醒應該是在早上 `08:00`。

會這樣推測是因為在我 1/26 又設定下次校正時間是 1/27 `10:00` 後，它雖然沒在 `10:00` 後我打開電腦時有任何反應，但它在今天 1/28 早上打開時突然提示要校正了，現在溫哥華時間還在 1/27 的下午，如果它真的是按溫哥華時間來，那麼溫哥華的早上 `10:00` 台北是 `02:00` 我在睡覺，當然沒開蓋，醒來後開蓋就提示，那就 make sense 了。

然後再自問自答，跑校正時有沒打開終端機並沒影響。

js4jiang5, 2025-01-29:
我過年後再看。

js4jiang5, 2025-01-30:
我目前只有在執行 dailylog 的當下順便確認 calibration 是否被跳過，如果被跳過就會執行。所以10:00被跳過，10:30 打開並不會馬上執行，而是到隔天執行 dailylog 時才會執行。
因為 calibration 很久才做一次，所以我之前覺得隔天再補做差不了多少。
要改成比較頻繁的做確認也是可以，下一版再改吧。

js4jiang5, 2025-01-30:
我以更新至 v2.0.19。請執行 "battery update"，更新後的版本應該會立即補做校正。

raymao9, 2025-01-30:
> 我以更新至 v2.0.19。請執行 "battery update"，更新後的版本應該會立即補做校正。

辛苦了，馬上更新~

closed 2025-02-17 by js4jiang5