js4jiang5/BatteryOptimizer_for_Mac #23: Battery status percentage and capacity different from system and other apps

teqtic, 2025-01-28:
**Your macbook model and year:**
2021 16" M1 Max MacBook Pro

**Your version number: (run "battery version" to find out)**
v2.0.18

**Your battery status: (run "battery status" to find out)**
01/28/25-14:34:14 - Battery at 71.0%, 11.816V, 30.9°C, discharging
01/28/25-14:34:14 - Battery health 99.1%, Cycle 38
01/28/25-14:34:14 - Your battery is currently being maintained at 80% with sailing to 75%
01/28/25-14:34:14 - You haven't scheduled calibration yet

**What is the issue?**
The battery percentage as reported by "battery status" (above) is 4% lower than that reported by the system or iStatMenus.  The battery health is also different. This reports it as 99% whereas the system reports it as 94%. 

Why is this?

**Error logs: (run "battery logs" to find out)**
No errors.

**Side notes:** 
During install I noticed it installs sleepwatcher, but running "brew services info sleepwatcher" shows it is not loaded or running. When is it used? 

I switched from the main "battery" project by actuallymentor because I noticed it would still charge over the limit if the MacBook was already sleeping when I plugged it in. Does this fork fix that?

Thanks for working on this!

js4jiang5, 2025-01-30:
The battery percentage and health provided by macOS is higher than real one, meaning they're too optimistic. The values I provided is from kernel directly. They're the same with coconutBattery and another famous app AlDente. 
Don't have to be too concerned about 81%, 82%, which is a couple percents higher than your target so long as the charging is stopped as expected. These values are controlled by MAC and inevidible. Truth is they varies day by day.

teqtic, 2025-01-30:
What's interesting is that the difference between 'battery' reported values and MacOS reported values only happens during discharge. While charging, they appear to be much closer to one another. 

How is the battery health calculated? it seems to be jumping all over the place (see log after calibration in first post).

js4jiang5, 2025-01-30:
Health is max capacity relative to design capacity.

teqtic, 2025-01-30:
Yes I know what its definition is. The reported value is changing so I guess the measured capacity is changing. Maybe I'm being lazy and should look through the code myself for these answers.

closed 2025-02-17 by js4jiang5