js4jiang5/BatteryOptimizer_for_Mac #24: Questions after first install

teqtic, 2025-01-29:
I initially had this in the bug report I opened another issue for, but had too many questions, so put them all here:

1) I switched from the main "battery" project by actuallymentor because I noticed it would still charge over the limit if the MacBook was already sleeping when I plugged it in. Does this fork fix that?

2) The main fix I'm interested in is "Battery health deteriorate due to simultaneous charge and discharge". Is that because the original battery project did not have a sail percentage? So it would constantly re-engage charging to keep it at the set level?

3) During install I noticed it installs sleepwatcher, but running "brew services info sleepwatcher" shows it is not loaded or running. When is it used?

js4jiang5, 2025-01-29:
I'll get back to you after lunar new year vacation.

SangTruongTan, 2025-01-30:
For the first question, I didn't visit the original source for months and someone implements shits into the project. Because I have a fork of the project, which works out of the box for sure.

js4jiang5, 2025-01-30:
> I initially had this in the bug report I opened another issue for, but had too many questions, so put them all here:
>
> 1. I switched from the main "battery" project by actuallymentor because I [...]

How about you try it to find out and let me know?

> 2. The main fix I'm interested in is "Battery health deteriorate due to simultaneous charge and discharge". Is that because the original battery project did not have a sail percentage? So it would constantly re-engage charging to keep it at the set level?

I think it is just a bug. Since it's not reasonable, no one knows what it will do to your MAC, and that's why I fixed it to avoid this twilight zone.

> 3. During install I noticed it installs sleepwatcher, but running "brew services info sleepwatcher" shows it is not loaded or running. When is it used?

sleepwatcher is a third-party app that I used to control MAC before it enters sleep. It is launched when you boot your macbook. You may use command "which sleepwatcher" to check if it is successfully installed. If not, nothing will do before sleep. If yes, it will stop charging at 80% during sleeping.

closed 2025-02-17 by js4jiang5