js4jiang5/BatteryOptimizer_for_Mac #29: Error msg while show battery status

rickie37, 2025-02-17:
**Your macbook model and year:**
rickie@Rickies-M ~ % system_profiler SPHardwareDataType
Hardware:

    Hardware Overview:

      Model Name: MacBook Air
      Model Identifier: MacBookAir10,1
      Model Number: Z127000YTTA/A
      Chip: Apple M1
      Total Number of Cores: 8 (4 performance and 4 efficiency)
      Memory: 16 GB
      System Firmware Version: 11881.81.2
      OS Loader Version: 11881.81.2

**Your version number: (run "battery version" to find out)**
rickie@Rickies-M ~ % battery version
v2.0.20

**Your battery status: (run "battery status" to find out)**
rickie@Rickies-M ~ % battery status

02/17/25-09:35:39 - Battery at 21.8%, 11.217V, 26.8°C, charging
02/17/25-09:35:39 - Battery health 93.2%, Cycle 59
02/17/25-09:35:39 - Your battery is currently being maintained at 90% with sailing to 70%
02/17/25-09:35:39 - Schedule calibration on MON every 2 week at 9:30
/usr/local/bin/battery: line 366: [[: 08: value too great for base (error token is "08")
02/17/25-09:35:39 - Next calibration date is 2025/03/03

**What is the issue?**
Error msg while show battery status.

**Error logs: (run "battery logs" to find out)**
/usr/local/bin/battery: line 366: [[: 08: value too great for base (error token is "08")

rickie37, 2025-02-17:
[[ will treat numbers starting with 0 as octal values, and 08 is not a valid octal number. The same issue is likely to occur in week 9 as well (09).

rickie37, 2025-02-17:
I have changed `date +%V` to $((10#$(date +%V))) in line 366 as a temporary solution.

rickie37, 2025-02-17:
I have changed `date +%V` to $((10##$(date +%V))) in line 366 as a temporary solution.

js4jiang5, 2025-02-17:
> I have changed `date +%V` to .$((10#$(date +%V))) in line 366 as a temporary solution.

Thank you for pointing out the bug and providing the solution. I'll fix it in v2.0.21.

closed 2025-02-19 by js4jiang5