js4jiang5/BatteryOptimizer_for_Mac #30: Script Editor Notification, First install

samuel-nitzsche-cariad, 2025-02-17:
**Your macbook model and year:**
MacBook Pro 16" 2024 (M4 Max)

**Your version number: (run "battery version" to find out)**
v2.0.21

**Your battery status: (run "battery status" to find out)**

**What is the issue?**

**Error logs: (run "battery logs" to find out)**

js4jiang5, 2025-02-17:
Pushing a dummy notification cannot make the app appear in the system notification settings list. The user must enable the settings in advance in order to receive notifications.

js4jiang5, 2025-02-18:
> I know that the user has to enable the notifications in order to recieve them but if you never got an notification from the Script Editor then it is not even in the list so you can't change it

I didn't get it because in my case the Script Editor was in the list without doing anything. Now I know what you mean. I'll add a dummy notification in setup.sh. Thank you.

closed 2025-02-19 by js4jiang5

jackblackevo, 2025-06-05:
I had the same issue. I had never run the Script Editor app on my MacBook before, so Script Editor wasn't listed under **System Settings > Notifications > Applications**.

My solution was to manually open the Script Editor app, then create a new document, enter and run the following script:

```applescript
display notification "Hello World!"
```

This triggered a system notification asking whether to allow Script Editor to send notifications.