js4jiang5/BatteryOptimizer_for_Mac #31: Continues to charge when the system turns off

emresaracoglu, 2025-02-17:
**Macbook M4 Pro, 2024 Nov**

**Your version number: 2.0.21**

**emre@Emre-MacBook-Pro ~ % battery status

02/18/25-00:09:44 - Battery at 80.0%, 12.382V, 23.1°C, no charging
02/18/25-00:09:44 - Battery health 101.8%, Cycle 3
02/18/25-00:09:44 - Your battery is currently being maintained at 80% with sailing to 70%
02/18/25-00:09:44 - Schedule calibration on WED every 2 week at 10:50
02/18/25-00:09:44 - Next calibration date is 2025/02/19

emre@Emre-MacBook-Pro ~ % **

Hello, When I turn off the computer, it keeps charging. Is it possible to remain at a set level?

js4jiang5, 2025-02-17:
It depends on your macOS. If macOS is Sequoia (macOS 15), that feature has been removed by Apple, so there's nothing we can do right now unless Apple change their mind in the future. However, turning off macbook frequently is not a good idea. Every reboot is a burden to the macbook. Personally I keep it always on to maintain the best efficiency. Shutdown and reboot is only necessary when something goes wring.

emresaracoglu, 2025-02-18:
Hi @js4jiang5 , I used to shut down my ThinkPad daily, but since this is my first time using a MacBook, I’ve decided to follow your suggestion and use sleep or deep sleep mode instead. However, if I don’t use it for a few days, I’ll shut it down.
Thank you for your help

js4jiang5, 2025-02-18:
> Hi [@js4jiang5](https://github.com/js4jiang5) , I used to shut down my ThinkPad daily, but since this is my first time using a MacBook, I’ve decided to follow your suggestion and use sleep or deep sleep mode instead. However, if I don’t use it for a few days, I’ll shut it down. Thank you for your help
> 
> Edit // Should the ‘_Optimized Battery Charging_’ option be enabled or disabled when using ‘BatteryOptimizer_for_MAC’?

It's better to disable that option.

closed 2025-02-18 by emresaracoglu