js4jiang5/BatteryOptimizer_for_Mac #32: Encountering issues related to battery.plist and Script Editor Notification

nush-cc, 2025-02-22:
**Your macbook model and year:**
Macbook air m2 (2022)

**Your macOS: (run "sw_vers" to find out)**
15.3.1

**Your version number: (run "battery version" to find out)**
v2.0.21

**Your battery status: (run "battery status" to find out)**

02/22/25-16:52:58 - Battery at 95.5%, 12.663V, 23.3°C, discharging
02/22/25-16:52:58 - Battery health 90.7%, Cycle 61
02/22/25-16:52:58 - Battery maintain is suspended
02/22/25-16:52:58 - You haven't scheduled calibration yet

**What is the issue?**
After completing the download via curl, I did not find the “Script Editor” notification field in the notification bar. I did see issue #30!

Additionally, when I run battery maintain 80, I get the error message:
/Users/user/Library/LaunchAgents/battery.plist: Permission denied,
but it seems that the discharge function is working normally.

Please let me know if I made any mistakes or if there are any modifications I need to make. Thank you!

**Error logs: (run "battery logs" to find out)**
02/22/25-16:47:38 - Killing old maintain process at
02/22/25-16:47:38 - New process ID 3621
02/22/25-16:47:38 - Writing new setting 80 to maintain_percentage
02/22/25-16:47:38 - Starting battery maintenance at 80% with sailing to 75%
02/22/25-16:47:38 - Not triggering discharge as it is not requested
02/22/25-16:47:38 - Daemon does not yet exist, creating daemon file at /Users/user/Library/LaunchAgents/battery.plist
02/22/25-16:47:38 - Charging to and maintaining at 80% from 96%
02/22/25-16:47:38 - 💡 Setting magsafe color to green
02/22/25-16:47:38 - Stop charge above 80
02/22/25-16:47:38 - 🔌🪫 Disabling battery charging
02/22/25-16:50:08 - Killing old maintain process at 3621 active
02/22/25-16:50:08 - New process ID 4296
02/22/25-16:50:08 - Writing new setting 80 to maintain_percentage
02/22/25-16:50:08 - Starting battery maintenance at 80% with sailing to 75%
02/22/25-16:50:08 - Not triggering discharge as it is not requested
02/22/25-16:50:08 - Daemon does not yet exist, creating daemon file at /Users/user/Library/LaunchAgents/battery.plist
02/22/25-16:50:08 - Charging to and maintaining at 80% from 96%
02/22/25-16:50:11 - ack battery maintain suspend

js4jiang5, 2025-02-22:
It seems you don't have access right to the folder /Users/user/Library/LaunchAgents/. 
It usually occurs when the macbook belongs to your company. In that case you'll need to ask the IT of your company to grant the access right to that folder. 
If you're the owner of the macbook, make sure the account type is "Administrator".

nush-cc, 2025-02-23:
Thank you for your reply. I am the owner of the computer and have confirmed that the currently logged-in account is an administrator. I checked the path (Users/user/Library/LaunchAgents/) and found that there is no battery.plist, only battery_shutdown.plist. I have also tried reinstalling multiple times and restarting the computer, but the issue remains. Thank you.

js4jiang5, 2025-02-23:
please run "ls -al ~/Library/LaunchAgents/battery_shutdown.plist"

nush-cc, 2025-02-23:
I got this response :
-rw-r--r--@ 1 root  staff  386  2 22 16:47 /Users/user/Library/LaunchAgents/battery_shutdown.plist

js4jiang5, 2025-02-23:
> I got this response : -rw-r--r--@ 1 root staff 386 2 22 16:47 /Users/user/Library/LaunchAgents/battery_shutdown.plist

Please run "sudo battery maintain 80 75". 
Enter password if got requested.

nush-cc, 2025-02-23:
I received this response, but the function seems to be working correctly.

Could not enable service: 125: Domain does not support specified action

02/23/25-13:33:19 - Battery at 79.5%, 12.144V, 24.0°C, discharging
02/23/25-13:33:19 - Battery health 90.0%, Cycle 61
02/23/25-13:33:19 - Your battery is currently being maintained at 80% with sailing to 75%
Could not print domain: 125: Domain does not support specified action
Could not print domain: 125: Domain does not support specified action
02/23/25-13:33:19 - You haven't scheduled calibration yet

js4jiang5, 2025-02-23:
Although tun function work, but battery.plist is meant to enable this app when you reboot macbook. There's still something wrong with your access right.
Please run 
"echo $USER" and 
"echo $HOME" and 
"echo $(id -u $USER)"

nush-cc, 2025-02-23:
╰─❯ echo $USER
nush

╰─❯ echo $HOME
/Users/nush

╰─❯ echo $(id -u $USER)
501

The reason I mentioned “Users/user” above is that I manually changed “nush” to “user.” In reality, it is “Users/nush.” Thank you.

js4jiang5, 2025-02-23:
Please run 
launchctl print gui/$(id -u $USER) | grep "=> enabled"

nush-cc, 2025-02-23:
I got this response, thank you.

╰─❯ launchctl print gui/$(id -u $USER) | grep "=> enabled"
		"com.docker.helper" => enabled
		"net.pulsesecure.WebLoginService" => enabled
		"net.pulsesecure.CertService" => enabled
		"net.pulsesecure.pulseagent" => enabled
		"com.battery_shutdown.app" => enabled
		"com.battery.app" => enabled
		"com.logi.cp-dev-mgr" => enabled

js4jiang5, 2025-02-23:
Please run
id -Gn nush | grep 'admin'

then run
battery status

nush-cc, 2025-02-23:
I got those responses, thank you.

╰─❯ id -Gn nush | grep 'admin'
staff everyone localaccounts _appserverusr admin _appserveradm _appstore _lpadmin _lpoperator _developer _analyticsusers com.apple.access_ftp com.apple.access_screensharing com.apple.access_ssh com.apple.access_remote_ae com.apple.sharepoint.group.1

╰─❯ battery status

02/23/25-14:28:28 - Battery at 68.0%, 11.788V, 27.6°C, discharging
02/23/25-14:28:28 - Battery health 90.5%, Cycle 61
02/23/25-14:28:28 - Battery maintain is not running
02/23/25-14:28:28 - You haven't scheduled calibration yet

js4jiang5, 2025-02-23:
For some reason your battery app is not running. Did you stop it?
Please run "battery maintain 80 75" again.

nush-cc, 2025-02-23:
Wow! It suddenly stopped reporting errors. However, I didn’t manually pause the battery maintenance just now. Should I reboot my MacBook to ensure that it works properly? Thank you so much!

╰─❯ battery maintain 80 75

02/23/25-14:33:08 - Battery at 67.1%, 11.798V, 27.7°C, discharging
02/23/25-14:33:08 - Battery health 90.7%, Cycle 61
02/23/25-14:33:08 - Your battery is currently being maintained at 80% with sailing to 75%
02/23/25-14:33:08 - You haven't scheduled calibration yet

js4jiang5, 2025-02-23:
It's still abnormal because it keeps discharging.
Did you unplug the power source?
If it's discharging due to unplug, please reboot your macbook.
After reboot succesfully, run "battery status" again.

nush-cc, 2025-02-23:
Yes, the “discharge” status is displayed because I am not currently connected to a power source. I restarted my computer, and the battery status shows that my battery maintenance is still functioning. Running battery maintain 80 75 again did not result in the same error as before. Thank you very much for your help.

closed 2025-02-23 by nush-cc