js4jiang5/BatteryOptimizer_for_Mac #34: 溫控功能

iversonshao, 2025-03-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
之前看到AIdente pro 他是可以控制電池溫度的 高低溫環境下也是會影響電池性能的對吧
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
所以不知道能不能有機會看到未來版本有溫度控制的功能呢？
或是偵測太高的時候 幫忙降溫 或是過熱保護等相關功能
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

js4jiang5, 2025-03-02:
Aldente 並沒有溫度控制的功能，它只是在溫度超過某個指定值時停止充電。

我目前並沒有打算這麼做，因為充電並不是造成溫度過高的主因，而且 Apple 自身已經有做智慧充電，根據實際狀況控制充電電流大小，所以實際的高速充電只有很短的時間。我不希望介入 Apple 對於充電速度控制的原有機制。

不同使用者對於自己的 Macbook 有不同的控制與設定，我希望這個 app 只是輔助，避免擅自介入修改使用者的設定，因此像 Clamshell mode 仍可執行放電或校正的功能，雖然做的到，但我一直沒有加入。因為可能因使用者使用不當而出問題，一旦出問題，使用者自己可能無法解決還得尋求幫助或解除安裝。而且，校正時必須打開螢幕，剛好可以用這個動作來控制校正開始的時間，或維持不開跳過此次校正，我覺得很不錯。

iversonshao, 2025-03-03:
謝謝你 原來你是這樣思考的 這樣我了解了

closed 2025-03-05 by js4jiang5