js4jiang5/BatteryOptimizer_for_Mac #35: Battery maintain is NOT getting back after reboot

SangTruongTan, 2025-03-07:
**Your macbook model and year:**
2021

**Your macOS: (run "sw_vers" to find out)**
macOS Sequoia 15.4

**Your version number: (run "battery version" to find out)**
v2.0.18

**Your battery status: (run "battery status" to find out)**
Battery at 55.4%, 11.593V, 32.7°C, discharging

**What is the issue?**
Battery maintain is NOT getting back after reboot
That might be helpful to note that the battery level is hiden when setting the maintain level.
03/07/25-08:34:22 - Battery at 55.8%, 12.513V, 34.0°C, charging
03/07/25-08:34:22 - Battery health 97.8%, Cycle 49
03/07/25-08:34:22 - Your battery is currently being maintained at <----- **at nothing here.**
03/07/25-08:34:22 - You haven't scheduled calibration yet

**Error logs: (run "battery logs" to find out)**
👾 Battery CLI logs:

02/04/25-08:39:21 - Daemon already exists, checking for differences
02/04/25-08:39:21 - Starting battery maintenance at 50% with sailing to 45%
02/04/25-08:39:21 - Not triggering discharge as it is not requested
02/04/25-08:39:22 - Charging to and maintaining at 50% from 57%
02/04/25-08:39:22 - Stop charge above 50
02/04/25-08:39:22 - 🔌🪫 Disabling battery charging
02/04/25-08:39:38 - ack battery maintain suspend
02/04/25-09:48:56 - 🔼🪫 Disabling battery discharging
02/04/25-09:48:57 - ack battery maintain recover
02/04/25-21:27:12 - 💡 Setting magsafe color to none
02/04/25-21:58:17 - Charge below 45
02/04/25-21:58:17 - 🔼🪫 Disabling battery discharging
02/04/25-21:58:18 - 🔌🔋 Enabling battery charging
02/04/25-23:48:05 - 💡 Setting magsafe color to green
02/05/25-00:04:34 - 💡 Setting magsafe color to none
02/05/25-01:00:45 - 💡 Setting magsafe color to green
02/05/25-01:00:50 - 💡 Setting magsafe color to none
02/05/25-04:50:48 - 💡 Setting magsafe color to green
02/05/25-05:05:04 - 💡 Setting magsafe color to none
02/05/25-07:14:37 - Stop charge above 50
02/05/25-07:14:37 - 🔌🪫 Disabling battery charging
02/05/25-07:31:31 - 💡 Setting magsafe color to green
02/07/25-16:34:07 - 💡 Setting magsafe color to none
02/07/25-16:35:57 - 💡 Setting magsafe color to green
02/08/25-20:04:19 - 💡 Setting magsafe color to none
02/09/25-01:08:44 - 💡 Setting magsafe color to green
02/09/25-02:00:39 - 💡 Setting magsafe color to none
curl: (6) Could not resolve host: raw.githubusercontent.com
curl: (6) Could not resolve host: raw.githubusercontent.com
02/09/25-10:32:32 - Charge below 45
02/09/25-10:32:32 - 🔼🪫 Disabling battery discharging
02/09/25-10:32:33 - 🔌🔋 Enabling battery charging
02/09/25-15:59:21 - 💡 Setting magsafe color to green
02/09/25-16:24:41 - 💡 Setting magsafe color to none
02/09/25-18:18:56 - 💡 Setting magsafe color to green
02/09/25-18:23:10 - 💡 Setting magsafe color to none
02/09/25-20:19:33 - 💡 Setting magsafe color to green
02/09/25-20:19:38 - 💡 Setting magsafe color to none
02/09/25-22:01:09 - 💡 Setting magsafe color to none
02/09/25-22:12:08 - Stop charge above 50
02/09/25-22:12:08 - 🔌🪫 Disabling battery charging
02/09/25-22:13:13 - 💡 Setting magsafe color to green
02/10/25-19:56:12 - 💡 Setting magsafe color to none
02/10/25-20:03:27 - 💡 Setting magsafe color to green
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
02/12/25-19:13:15 - Debug trail. User: sangt, config folder: /Users/sangt/.battery, logfile: /Users/sangt/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
[... 125 lines cut ...]
Battery status:
{"percentage":"51","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"50","battery_state":"51% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 8432
drwxr-xr-x@  9 sangt  staff   288B 30 Jan 08:45 .
drwxr-x---+ 68 sangt  staff   2.1K  7 Mar 08:38 ..
-rwxr-xr-x@  1 sangt  staff   4.1M  7 Mar 08:39 battery.log
-rw-r--r--@  1 sangt  staff    15B  7 Mar 08:34 battery.pid
-rw-r--r--@  1 sangt  staff   179B 30 Jan 08:45 calibrate.log
-rw-r--r--@  1 sangt  staff   2.5K  7 Mar 00:02 daily.log
-rw-r--r--@  1 sangt  staff   5.4K 29 Jan 07:11 gui.log
-rw-r--r--@  1 sangt  staff     3B 29 Jan 08:23 maintain.percentage
-rw-r--r--@  1 sangt  staff    25B  7 Mar 08:34 sig.pid

⚙️	Battery data:

03/07/25-08:39:51 - Battery at 55.2%, 11.561V, 32.5°C, discharging
03/07/25-08:39:51 - Battery health 97.8%, Cycle 49
03/07/25-08:39:51 - Battery maintain is suspended
03/07/25-08:39:51 - You haven't scheduled calibration yet

Battery CLI utility v2.0.1

js4jiang5, 2025-03-07:
For some unknown reason, your configuration file is missing. Besides, it seems that you directly upgrade from the "battery" app of original version with GUI.
Please run "battery uninstall" to remove old version completely, and then install from scratch based on README of my github. Everything should be back to normal after reinstall. Good luck.

SangTruongTan, 2025-03-08:
You made my day! After reinstalling the battery tool, it works out of the box.

closed 2025-03-08 by SangTruongTan