js4jiang5/BatteryOptimizer_for_Mac #36: Battery charged above 80% limit when laptop left closed and charging over night

princefl, 2025-03-09:
**Your macbook model and year:** M1 2020

**Your macOS: 14.5

**Your version number: 2.0.21

**Your battery status: (run "battery status" to find out)**03/08/25-23:42:45 - Battery at 88.1%, 12.442V, 24.4°C, discharging
03/08/25-23:42:46 - Battery health 94.2%, Cycle 60
03/08/25-23:42:46 - Your battery is currently being maintained at 80% with sailing to 75%

**What is the issue?** Battery charged above 80% when leaving MacBook plugged in overnight, macOS says I'm at 93% and terminal says I'm at 88.1%

**Error logs: (run "battery logs" to find out)**03/07/25-23:25:35 - Killing old maintain process at 
03/07/25-23:25:35 - New process ID 27738
03/07/25-23:25:35 - Writing new setting 80 to maintain_percentage
03/07/25-23:25:35 - Daemon does not yet exist, creating daemon file at /Users/princelawrence/Library/LaunchAgents/battery.plist
03/07/25-23:25:35 - Starting battery maintenance at 80% with sailing to 75% 
03/07/25-23:25:35 - Not triggering discharge as it is not requested
03/07/25-23:25:35 - Charging to and maintaining at 80% from 35%
03/07/25-23:34:03 - Killing old maintain process at 27738 active
03/07/25-23:34:03 - New process ID 29516
03/07/25-23:34:03 - Writing new setting 80 to maintain_percentage
03/07/25-23:34:03 - Daemon already exists, checking for differences
03/07/25-23:34:03 - Starting battery maintenance at 80% with sailing to 75% 
03/07/25-23:34:03 - Not triggering discharge as it is not requested
03/07/25-23:34:03 - Charging to and maintaining at 80% from 33%
03/08/25-02:09:39 - Stop charge above 80
03/08/25-02:09:39 - 🔌🪫 Disabling battery charging

Sorry if dumb question or obvious reason its not working, I am not a dev by any means so I don't understand this stuff.

js4jiang5, 2025-03-09:
The battery was 88.1%. When you ran "battery maintain 80", a window should have pop-up asking if you want to discharge to 80%. You answered no, so it was kept 88.1%.
You may run "battery maintain 80" again, and answer "yes" to discharge immediately to 80%.

princefl, 2025-03-09:
> The battery was 88.1%. When you ran "battery maintain 80", a window should have pop-up asking if you want to discharge to 80%. You answered no, so it was kept 88.1%. You may run "battery maintain 80" again, and answer "yes" to discharge immediately to 80%.

I was at 33% last night when I ran battery maintain 80. woke up this morning and it was at 88.1%

js4jiang5, 2025-03-09:
> I was at 33% last night when I ran battery maintain 80. woke up this morning and it was at 88.1%

Please run "which sleepwatcher" and see what the result is.

princefl, 2025-03-09:
> Please run "which sleepwatcher" and see what the result is.

it says "sleep watcher not found" how do I fix this? thanks for helping!

js4jiang5, 2025-03-09:
It means either homebrew or sleepwatcher install failed when you install this app.
When you close the lid, your macbook enter sleep mode.
Without sleepwatcher, it will continue to charge if the status is charging when you close the lid.
Please run "which brew".
If homebrew is not found, either, you'll need to install homebrew by running
"curl -s https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh | bash"

if homebrew exist, install sleepwatcher by running
"HOMEBREW_NO_INSTALL_FROM_API=1 brew reinstall sleepwatcher"

princefl, 2025-03-09:
Thank you, I actually ended up uninstalling and rerunning the installation script before seeing your reply and sleepwatcher installed this time. Thanks for the help, new to Macs so sorry if obvious questions.

closed 2025-03-09 by princefl

js4jiang5, 2025-04-11:
> hi [@js4jiang5](https://github.com/js4jiang5)
> 
> Thanks for this great tool save my battery.
> 
> I have a question though: why use HOMEBREW_NO_INSTALL_FROM_AP=1 to install sleepwatcher? It ends up downloading an extra 1GB of data onto Apple's hard-drive-made-of-gold 😛
> 
> thanks again!

I didn't know about 1GB. I remember in some case the installation might fail and adding the setting can avoid the problem.

brookqin, 2025-04-12:
https://docs.brew.sh/Installation#default-tap-cloning

Using `HOMEBREW_NO_INSTALL_FROM_API=1` will make brew clone the entire Homebrew/homebrew-core repository.

The `brew doctor` command may warn you that cloning the core repository could potentially cause conflicts.

js4jiang5, 2025-04-12:
> https://docs.brew.sh/Installation#default-tap-cloning
> 
> Using `HOMEBREW_NO_INSTALL_FROM_API=1` will make brew clone the entire Homebrew/homebrew-core repository.
> 
> The `brew doctor` command may warn you that cloning the core repository could potentially cause conflicts.

I remember I ran into the same problem [here](https://github.com/orgs/Homebrew/discussions/4231) and eventually used his solution without thinking too much.