js4jiang5/BatteryOptimizer_for_Mac #38: Battery calibration stocked at 99% charging

leonhsieh, 2025-03-16:
**Your macbook model and year:** 
2023 M3 MBP

**Your macOS: (run "sw_vers" to find out)**
 14.1

**Your version number: (run "battery version" to find out)** 
v2.0.21

**Your battery status: (run "battery status" to find out)**
03/16/25-22:52:14 - Battery at 99% (target 100%)

**What is the issue?**
Battery charging process stocked at 99% for over 30 minutes with orange mega safe light.

**Error logs: (run "battery logs" to find out)**

-MacBook-Pro ~ % battery calibrate
03/16/25-20:20:37 - Calibration: Please open macbook lid and connect AC to start calibration
03/16/25-20:20:43 - Calibration: Calibration has started! Start discharging to 15%
03/16/25-20:20:44 - Suspend in 5 secs, wait ..
03/16/25-20:20:45 - Battery maintain is suspended
03/16/25-20:20:45 - Discharging to 15% from 26%
03/16/25-20:20:45 - 🔌🪫 Disabling battery charging
03/16/25-20:20:46 - 🔽🪫 Enabling battery discharging
03/16/25-20:20:47 - 💡 Setting magsafe color to none
03/16/25-20:24:47 - Battery at 25% (target 15%)
03/16/25-20:28:48 - Battery at 24% (target 15%)
03/16/25-20:32:48 - Battery at 23% (target 15%)
03/16/25-20:40:49 - Battery at 22% (target 15%)
03/16/25-20:53:49 - Battery at 21% (target 15%)
03/16/25-20:59:50 - Battery at 20% (target 15%)
03/16/25-21:11:51 - Battery at 19% (target 15%)
03/16/25-21:17:51 - Battery at 18% (target 15%)
03/16/25-21:25:52 - Battery at 17% (target 15%)
03/16/25-21:30:52 - Battery at 16% (target 15%)
03/16/25-21:34:52 - 🔼🪫 Disabling battery discharging
03/16/25-21:34:58 - 💡 Setting magsafe color to green
03/16/25-21:34:58 - 電池已放電至 15%
03/16/25-21:34:58 - Calibration: Calibration has discharged to 15%. Start charging to 100%
03/16/25-21:34:59 - Charging to 100% from 15%
03/16/25-21:34:59 - 🔼🪫 Disabling battery discharging
03/16/25-21:35:00 - 🔌🔋 Enabling battery charging
03/16/25-21:35:01 - 💡 Setting magsafe color to orange
03/16/25-21:36:01 - Battery at 17% (target 100%)
03/16/25-21:37:01 - Battery at 19% (target 100%)
03/16/25-21:38:01 - Battery at 21% (target 100%)
03/16/25-21:39:01 - Battery at 22% (target 100%)
03/16/25-21:40:01 - Battery at 24% (target 100%)
03/16/25-21:41:01 - Battery at 25% (target 100%)
03/16/25-21:42:01 - Battery at 27% (target 100%)
03/16/25-21:43:02 - Battery at 29% (target 100%)
03/16/25-21:44:02 - Battery at 30% (target 100%)
03/16/25-21:45:02 - Battery at 32% (target 100%)
03/16/25-21:46:02 - Battery at 33% (target 100%)
03/16/25-21:47:02 - Battery at 35% (target 100%)
03/16/25-21:48:02 - Battery at 37% (target 100%)
^[[1;2D03/16/25-21:49:02 - Battery at 38% (target 100%)
03/16/25-21:50:02 - Battery at 40% (target 100%)
03/16/25-21:51:02 - Battery at 42% (target 100%)
03/16/25-21:52:02 - Battery at 44% (target 100%)
03/16/25-21:53:02 - Battery at 46% (target 100%)
03/16/25-21:54:02 - Battery at 47% (target 100%)
03/16/25-21:55:02 - Battery at 49% (target 100%)
03/16/25-21:56:02 - Battery at 51% (target 100%)
03/16/25-21:57:02 - Battery at 52% (target 100%)
03/16/25-21:58:02 - Battery at 54% (target 100%)
03/16/25-21:59:03 - Battery at 56% (target 100%)
03/16/25-22:00:03 - Battery at 57% (target 100%)
03/16/25-22:01:03 - Battery at 58% (target 100%)
03/16/25-22:02:03 - Battery at 60% (target 100%)
03/16/25-22:03:03 - Battery at 61% (target 100%)
03/16/25-22:04:03 - Battery at 63% (target 100%)
03/16/25-22:05:03 - Battery at 65% (target 100%)
03/16/25-22:06:03 - Battery at 66% (target 100%)
03/16/25-22:07:03 - Battery at 68% (target 100%)
03/16/25-22:08:03 - Battery at 69% (target 100%)
03/16/25-22:09:03 - Battery at 70% (target 100%)
03/16/25-22:10:03 - Battery at 71% (target 100%)
03/16/25-22:11:03 - Battery at 73% (target 100%)
03/16/25-22:12:03 - Battery at 74% (target 100%)
03/16/25-22:13:03 - Battery at 75% (target 100%)
03/16/25-22:14:03 - Battery at 76% (target 100%)
03/16/25-22:15:03 - Battery at 77% (target 100%)
03/16/25-22:16:03 - Battery at 78% (target 100%)
03/16/25-22:17:04 - Battery at 79% (target 100%)
03/16/25-22:18:04 - Battery at 80% (target 100%)
03/16/25-22:19:04 - Battery at 81% (target 100%)
03/16/25-22:21:04 - Battery at 82% (target 100%)
03/16/25-22:22:04 - Battery at 83% (target 100%)
03/16/25-22:24:04 - Battery at 84% (target 100%)
03/16/25-22:25:04 - Battery at 85% (target 100%)
03/16/25-22:27:04 - Battery at 86% (target 100%)
03/16/25-22:28:04 - Battery at 87% (target 100%)
03/16/25-22:30:04 - Battery at 88% (target 100%)
03/16/25-22:31:04 - Battery at 89% (target 100%)
03/16/25-22:33:04 - Battery at 90% (target 100%)
03/16/25-22:34:05 - Battery at 91% (target 100%)
03/16/25-22:36:05 - Battery at 92% (target 100%)
03/16/25-22:37:05 - Battery at 93% (target 100%)
03/16/25-22:39:05 - Battery at 94% (target 100%)
03/16/25-22:41:05 - Battery at 95% (target 100%)
03/16/25-22:43:05 - Battery at 96% (target 100%)
03/16/25-22:46:05 - Battery at 97% (target 100%)
03/16/25-22:48:27 - Battery at 98% (target 100%)
03/16/25-22:52:14 - Battery at 99% (target 100%)

leonhsieh, 2025-03-16:
Right.... so the log updated, but my terminal still running wired.
03/16/25-22:48:27 - Battery at 98% (target 100%)
03/16/25-22:52:14 - Battery at 99% (target 100%)
03/16/25-23:25:43 - 💡 Setting magsafe color to green
03/16/25-23:25:43 - 電池已充電至 100%
03/16/25-23:25:43 - Calibration: Calibration has charged to 100%. Waiting for one hour
03/16/25-23:25:43 - 💡 Setting magsafe color to green

js4jiang5, 2025-03-16:
That's normal.
Charging from 99 to 100% with a long time happens when the battery hasn't been calibrated for quite a while, resulting inaccurate battery capacity. The battery health usually improves after the callibration in this situation.

closed 2025-03-18 by js4jiang5