js4jiang5/BatteryOptimizer_for_Mac #40: Battery is charged above maintain %

SebastianDadeby, 2025-03-19:
**Your macbook model and year:**
MacBook Air M3 2024
**Your macOS: (run "sw_vers" to find out)**
ProductName:		macOS
ProductVersion:		15.0
BuildVersion:		24A335
**Your version number: (run "battery version" to find out)**
v2.0.21
**Your battery status: (run "battery status" to find out)**
03/19/25-12:50:19 - Battery at 85.0%, 12.763V, 25.1°C, no charging
03/19/25-12:50:19 - Battery health 99.4%, Cycle 85
03/19/25-12:50:19 - Your battery is currently being maintained at 80% with sailing to 75%
03/19/25-12:50:19 - You haven't scheduled calibration yet
**What is the issue?**
Maintain is set to 80%, but battery charges all the way to 85% before maintaining. Happens when the charger is disconnected so that the battery is drained below the maintenance level. When reconnected, the battery is charged 5-10% above the set maintenance threshold.
**Error logs: (run "battery logs" to find out)**
👾 Battery CLI logs:

03/17/25-08:23:03 - Killing old maintain process at 
03/17/25-08:23:03 - New process ID 8308
03/17/25-08:23:03 - Writing new setting 80 to maintain_percentage
03/17/25-08:23:03 - Starting battery maintenance at 80% with sailing to 75% 
03/17/25-08:23:03 - Not triggering discharge as it is not requested
03/17/25-08:23:03 - Daemon does not yet exist, creating daemon file at /Users/sdadeby/Library/LaunchAgents/battery.plist
03/17/25-08:23:03 - Charging to and maintaining at 80% from 95%
03/17/25-08:23:04 - Stop charge above 80
03/17/25-08:23:04 - 🔌🪫 Disabling battery charging
03/17/25-08:23:15 - ack battery maintain suspend
03/17/25-08:23:41 - 🔼🪫 Disabling battery discharging
03/17/25-08:23:42 - ack battery maintain recover
03/17/25-08:23:42 - 💡 Setting magsafe color to green
03/17/25-08:37:19 - ack battery maintain suspend
03/18/25-14:50:27 - 🔼🪫 Disabling battery discharging
03/18/25-14:50:28 - ack battery maintain recover
03/18/25-14:50:28 - Stop charge above 80
03/18/25-14:50:28 - 🔌🪫 Disabling battery charging
03/18/25-14:50:57 - ack battery maintain suspend
03/18/25-21:22:46 - Battery maintain is recovered because AC adapter is reconnected
03/18/25-21:22:46 - 💡 Setting magsafe color to none
03/19/25-08:02:08 - 💡 Setting magsafe color to green
03/19/25-08:03:20 - 💡 Setting magsafe color to none
03/19/25-08:25:19 - 💡 Setting magsafe color to none
03/19/25-08:41:21 - 💡 Setting magsafe color to green
03/19/25-09:05:33 - 💡 Setting magsafe color to none
03/19/25-09:35:08 - Charge below 75
03/19/25-09:35:08 - 🔼🪫 Disabling battery discharging
03/19/25-09:35:09 - 🔌🔋 Enabling battery charging
03/19/25-12:13:33 - Stop charge above 80
03/19/25-12:13:33 - 🔌🪫 Disabling battery charging
03/19/25-12:29:58 - 💡 Setting magsafe color to green

js4jiang5, 2025-03-19:
Please refer to issue #37

js4jiang5, 2025-03-19:
> In this case the battery level reported i by `battery` itself, not the MacOS battery indicator. As you can see on the `battery status` command, the battery level is reported to be 85% even though I've set it to maintain at 80%. This was after plugging the laptop back in at ~60% charge.

I guess it's a natural physical phenomenon that mac keeps charging for a short period of time after charging is disabled. You may check the README of [bclm](https://github.com/zackelia/bclm) and follow its recommendation. 

[image]

Truth is even Apple CPU Macs also have similar behavior.

closed 2025-03-21 by js4jiang5