js4jiang5/BatteryOptimizer_for_Mac #44: macOS 26 support

afawe3331, 2025-06-10:
Currently this app does not work on macOS 26.
Battery discharging does not work. LED function still seems to work

js4jiang5, 2025-06-10:
Thank you for the information.
Not only this app, the other similar battery charge limiter such as AlDente all failed to work in macOS 26 Tahoe beta.
There's nothing we can do right now.
If battery optimizer is very important to you, you'd better roll back to Sequoia.

SangTruongTan, 2025-06-12:
Did you try figuring out what's happening with macOS 26 Tahoe for now, @js4jiang5? I've just experienced the same issue on my MBP.

Macrumers1310, 2025-06-17:
> macOS 26 changed the SMC control for fan, temperature, and battery. This beta macOS is for developer only, not for customer users. If you're not a developer, don't upgrade to macOS 26.

Can you solve it? Save us T.T

peters0908, 2025-06-19:
AlDente 1.33.2 好像有找到在macOS 26限制充電的解法了，FYI

js4jiang5, 2025-07-01:
I just released a beta version using newly created "tahoe" branch.
If you're already using macOS 26 Tahoe, you may get the beta version with command "battery update beta tahoe"
Let me know if it works for you.

closed 2025-07-02 by afawe3331