js4jiang5/BatteryOptimizer_for_Mac #49: macOS Tahoe support not working on my MacBook Air m4

GerTeunis, 2025-07-06:
**Your macbook model and year:**
Macbook Air M4

**Your macOS: (run "sw_vers" to find out)**
macOS Tahoe 26.0 Beta (25A5295e)

**Your version number: (run "battery version" to find out)**
v2.0.23

**Your battery status: (run "battery status" to find out)**
07/06/25-10:14:56 - Battery at 99.7%, 13.245V, 26.2°C, discharging
07/06/25-10:14:56 - Battery health 102.5%, Cycle 6
07/06/25-10:14:56 - Your battery is currently being maintained at 70% with sailing to 65%
07/06/25-10:14:56 - You haven't scheduled calibration yet

**What is the issue?**
I had it discharged to 70%, plugged it in
My Macbook was charged to 100% sadly

**Error logs: (run "battery logs" to find out)**
`id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list

🖥️	Battery GUI logs:

tail: /Users/gerteunis/.battery/gui.log: No such file or directory

📁 Config folder details:

total 6192
drwxr-xr-x   8 gerteunis  staff   256B Jul  6 00:00 .
drwxr-x---+ 32 gerteunis  staff   1.0K Jul  6 10:00 ..
-rwxr-xr-x   1 gerteunis  staff   2.1M Jul  6 10:15 battery.log
-rwxr-xr-x   1 gerteunis  staff    13B Jul  5 23:03 battery.pid
-rw-r--r--   1 gerteunis  staff   186B Jul  6 00:00 config_battery
-rw-r--r--   1 gerteunis  staff   1.4K Jul  6 00:00 daily.log
-rw-r--r--   1 gerteunis  staff    14B Jul  5 20:42 sig.pid
-rw-r--r--   1 gerteunis  staff   232B Jul  5 23:03 sleepwatcher.log

⚙️	Battery data:

07/06/25-10:15:49 - Battery at 99.6%, 13.247V, 26.6°C, discharging
07/06/25-10:15:49 - Battery health 102.4%, Cycle 6
07/06/25-10:15:49 - Your battery is currently being maintained at 70% with sailing to 65%
07/06/25-10:15:49 - You haven't scheduled calibration yet

Battery CLI utility v2.0.23
`

js4jiang5, 2025-07-06:
Please reboot your Macbook and run "battery reinstall" and see if it works.

GerTeunis, 2025-07-06:
Rebooted, reinstalled as instructed. Didn'y change anything

```
gerteunis@Gers-MacBook-Air ~ % battery logs     
👾 Battery CLI logs:

07/06/25-10:21:44 - Killing old maintain process at 
07/06/25-10:21:44 - New process ID 21691
07/06/25-10:21:44 - Writing new setting 80 to maintain_percentage
07/06/25-10:21:44 - Starting battery maintenance at 80% with sailing to 75% 
07/06/25-10:21:44 - Not triggering discharge as it is not requested
07/06/25-10:21:44 - Daemon does not yet exist, creating daemon file at /Users/gerteunis/Library/LaunchAgents/battery.plist
07/06/25-10:21:44 - Charging to and maintaining at 80% from 99%
07/06/25-10:21:56 - ack battery maintain suspend
07/06/25-10:23:08 - Killing old maintain process at 21691 suspended
07/06/25-10:23:08 - New process ID 22525
07/06/25-10:23:08 - Writing new setting 70 to maintain_percentage
07/06/25-10:23:08 - Starting battery maintenance at 70% with sailing to 65% 
07/06/25-10:23:08 - Not triggering discharge as it is not requested
07/06/25-10:23:08 - Daemon already exists, checking for differences
07/06/25-10:23:08 - Charging to and maintaining at 70% from 99%
07/06/25-10:23:12 - ack battery maintain suspend
07/06/25-10:53:34 - Battery maintain is recovered because AC adapter is reconnected
07/06/25-10:53:35 - 💡 Setting magsafe color to none
07/06/25-11:42:49 - 💡 Setting magsafe color to none
07/06/25-13:23:44 - 💡 Setting magsafe color to none
07/06/25-13:31:35 - 💡 Setting magsafe color to none
07/06/25-14:22:36 - 💡 Setting magsafe color to none
07/06/25-14:31:58 - 💡 Setting magsafe color to none
07/06/25-14:34:26 - 💡 Setting magsafe color to none
07/06/25-15:51:42 - 💡 Setting magsafe color to none
07/06/25-17:03:37 - 💡 Setting magsafe color to none
07/06/25-18:04:39 - 💡 Setting magsafe color to none
07/06/25-18:43:39 - 💡 Setting magsafe color to none
07/06/25-20:53:00 - Debug trail. User: gerteunis, config folder: /Users/gerteunis/.battery, logfile: /Users/gerteunis/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
07/06/25-20:53:00 - Recovering maintenance percentage 70
07/06/25-20:53:00 - Starting battery maintenance at 70% with sailing to 65% 
07/06/25-20:53:00 - Not triggering discharge as it is not requested
07/06/25-20:53:00 - Charging to and maintaining at 70% from 68%
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
id: gerteunis: no such user
Could not print domain: 141: Reentrancy avoided
sysmon request failed with error: sysmond service not found
pgrep: Cannot get process list
07/06/25-20:54:30 - Debug trail. User: gerteunis, config folder: /Users/gerteunis/.battery, logfile: /Users/gerteunis/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
07/06/25-20:54:30 - Recovering maintenance percentage 70
07/06/25-20:54:30 - Starting battery maintenance at 70% with sailing to 65% 
07/06/25-20:54:30 - Not triggering discharge as it is not requested
07/06/25-20:54:30 - Charging to and maintaining at 70% from 68%
07/06/25-20:54:31 - 💡 Setting magsafe color to none
07/06/25-21:06:52 - Debug trail. User: gerteunis, config folder: /Users/gerteunis/.battery, logfile: /Users/gerteunis/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
07/06/25-21:06:52 - Recovering maintenance percentage 70
07/06/25-21:06:52 - Starting battery maintenance at 70% with sailing to 65% 
07/06/25-21:06:52 - Not triggering discharge as it is not requested
07/06/25-21:06:52 - Charging to and maintaining at 70% from 75%
07/06/25-21:07:12 - Killing old maintain process at 719 active
07/06/25-21:07:12 - New process ID 1331
07/06/25-21:07:12 - Writing new setting 80 to maintain_percentage
07/06/25-21:07:12 - Starting battery maintenance at 80% with sailing to 75% 
07/06/25-21:07:12 - Not triggering discharge as it is not requested
07/06/25-21:07:12 - Daemon already exists, checking for differences
07/06/25-21:07:12 - Charging to and maintaining at 80% from 75%

🖥️	Battery GUI logs:

tail: /Users/gerteunis/.battery/gui.log: No such file or directory

📁 Config folder details:

total 48
drwxr-xr-x   8 gerteunis  staff   256B Jul  6 21:07 .
drwxr-x---+ 32 gerteunis  staff   1.0K Jul  6 21:07 ..
-rwxr-xr-x   1 gerteunis  staff   3.8K Jul  6 21:12 battery.log
-rwxr-xr-x   1 gerteunis  staff    12B Jul  6 21:07 battery.pid
-rw-r--r--   1 gerteunis  staff   176B Jul  6 21:07 config_battery
-rw-r--r--   1 gerteunis  staff    69B Jul  6 10:21 daily.log
-rw-r--r--   1 gerteunis  staff    26B Jul  6 10:23 sig.pid
-rw-r--r--   1 gerteunis  staff   145B Jul  6 21:07 sleepwatcher.log

⚙️	Battery data:

07/06/25-21:12:55 - Battery at 79.0%, 12.871V, 26.3°C, charging
07/06/25-21:12:55 - Battery health 101.6%, Cycle 7
07/06/25-21:12:55 - Your battery is currently being maintained at 80% with sailing to 75%
07/06/25-21:12:55 - You haven't scheduled calibration yet

Battery CLI utility v2.0.23

```

Did change to way beyond the charge limit and info shows now:

gerteunis@Gers-MacBook-Air ~ % battery status

```
07/06/25-21:18:40 - Battery at 83.0%, 13.018V, 26.3°C, charging
07/06/25-21:18:40 - Battery health 101.6%, Cycle 7
07/06/25-21:18:40 - Your battery is currently being maintained at 80% with sailing to 75%
07/06/25-21:18:40 - You haven't scheduled calibration yet

```

Any ideas?

js4jiang5, 2025-07-07:
Please run the following commands and show your results 
1. smc -k CH0B -r
2. smc -k CH0K -r
3. smc -k CH0J -r
4. smc -k CHTE -r
5. battery visudo

GerTeunis, 2025-07-07:
I can see the current state correctly

```
gerteunis@Gers-MacBook-Air ~ % battery status

07/07/25-19:27:00 - Battery at 81.0%, 12.945V, 25.4°C, charging
07/07/25-19:27:00 - Battery health 101.4%, Cycle 7
07/07/25-19:27:00 - Your battery is currently being maintained at 80% with sailing to 75%
07/07/25-19:27:00 - You haven't scheduled calibration yet
```

During the "Discharge to 80%" yesterday it was able to set the LED correctly to off as well.
But currently it's not stopping the charge what already at 81%.

During discharge I saw 2 battery processes. After reinstall it did disable the discharge mode, not I see just one

```
ps aux | grep battery
gerteunis         2610   0.0  0.0 435304416   2368   ??  S     9:45PM   0:54.69 /bin/bash /usr/local/bin/battery maintain_synchronous 80

```

GerTeunis, 2025-07-07:
manually starting discharge it does indeed stop charging:

```
gerteunis@Gers-MacBook-Air ~ % battery discharge 80
07/07/25-19:30:59 - Suspend in 5 secs, wait ..
07/07/25-19:31:00 - Battery maintain is suspended
07/07/25-19:31:00 - Discharging to 80% from 84%
07/07/25-19:31:00 - 🔌🪫 Disabling battery charging
07/07/25-19:31:01 - 🔽🪫 Enabling battery discharging
07/07/25-19:31:02 - 💡 Setting magsafe color to none
```

With two processes:

```
gerteunis         2610   0.0  0.0 435304416   2416   ??  S     9:45PM   0:54.76 /bin/bash /usr/local/bin/battery maintain_synchronous 80
gerteunis        25324   0.0  0.0 435300464   1408 s001  S+    7:31PM   0:00.00 grep battery
gerteunis        25073   0.0  0.0 435304416   2752 s000  S+    7:30PM   0:00.02 /bin/bash /usr/local/bin/battery discharge 80
```

GerTeunis, 2025-07-07:
macOS icon says: charging. Led is green. But is IS charging.
Although battery says: not charging

```
battery status

07/07/25-20:01:22 - Battery at 79.9%, 12.611V, 29.7°C, no charging
07/07/25-20:01:22 - Battery health 101.4%, Cycle 7
07/07/25-20:01:22 - Your battery is currently being maintained at 80% with sailing to 75%
07/07/25-20:01:22 - You haven't scheduled calibration yet
```

GerTeunis, 2025-07-07:
Led is green, macOS says: charging. Battery says: not charging.

```
smc -k CH0B -r
CH0B [ ] no data

smc -k CH0K -r
CH0K [ ] no data

smc -k CH0J -r
CH0J [ui8 ] 0 (bytes 00)

smc -k CHTE -r
  CHTE  [ui32]  16777216 (bytes 01 00 00 00)
```

GerTeunis, 2025-07-08:
Keeps charging past the set mark

smc -k CHOJ -r
  CHOJ  [    ]  no data

smc -k CHTE -r
  CHTE  [ui32]  0 (bytes 00 00 00 00)

smc -k CHBI -r
  CHBI  [ui32]  2533425152 (bytes 97 01 00 00)

smc -k B0AC -r
  B0AC  [si16]  35841 (bytes 8c 01)

battery status
```
07/08/25-20:47:23 - Battery at 98.9%, 13.338V, 24.9°C, charging
07/08/25-20:47:23 - Battery health 100.2%, Cycle 8
07/08/25-20:47:23 - Your battery is currently being maintained at 70% with sailing to 65%
07/08/25-20:47:24 - You haven't scheduled calibration yet
```

GerTeunis, 2025-07-08:
I also get these in the battery.log file

```
07/08/25-19:42:33 - Charge below 65
07/08/25-19:42:33 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002c1
07/08/25-19:42:34 - 🔌🔋 Enabling battery charging
07/08/25-19:42:40 - Charge below 65
07/08/25-19:42:40 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002c1
07/08/25-19:42:41 - 🔌🔋 Enabling battery charging
07/08/25-19:42:47 - Charge below 65
07/08/25-19:42:47 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002c1
07/08/25-19:42:48 - 🔌🔋 Enabling battery charging
```

js4jiang5, 2025-07-08:
It seems that the "stop charging" command is not effective in your Mac. Please run

1. sudo smc -k CHTE -w 01000000 #this is the command to stop charging
2. smc -k CHTE -r # read the value to check if it is written correctly

js4jiang5, 2025-07-08:
> ```
> gerteunis@Gers-MacBook-Air .battery % sudo smc -k CHTE -w 01000000
> gerteunis@Gers-MacBook-Air .battery % battery status              
> 
> 07/08/25-22:15:44 - Battery at 98.9%, 13.224V, 22.1°C, no charging
> 07/08/25-22:15:44 - Battery health 101.1%, Cycle 8
> 07/08/25-22:15:44 - Your battery is currently being maintained at 80% with sailing to 75%
> 07/08/25-22:15:44 - You haven't scheduled calibration yet
> 
> gerteunis@Gers-MacBook-Air .battery % smc -k CHTE -r
>   CHTE  [ui32]  16777216 (bytes 01 00 00 00)
> ```

It works. Please disable built-in macOS battery optimization to see if that caused conflict.
System Settings > Battery > Battery Health > click the info icon > toggle off "Optimize Battery Charging".

GerTeunis, 2025-07-09:
Thanks! Works like a charm!

closed 2025-07-09 by GerTeunis