js4jiang5/BatteryOptimizer_for_Mac #59: Every time I close the MacBook lid, then open it again, I get a sudo prompt, which is annoying

SwiftWinds, 2025-08-31:
Every time I close the MacBook lid, then open it again, I get this sudo prompt:

[image]

Which is very annoying. When running `ps aux | grep sudo`, I get:

```
mateowan         91411   0.0  0.0 410724448   1424 s000  S+   12:39AM   0:00.00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn --exclude-dir=.idea --exclude-dir=.tox --exclude-dir=.venv --exclude-dir=venv sudo
root             91295   0.0  0.0 410774304  11344   ??  S    12:38AM   0:00.02 sudo smc -k CHWA -w 00
```

Is there any way to prevent this prompt from keeping on appearing?

js4jiang5, 2025-08-31:
Your macOS version is 15.6, so the SMC key CHWA shouldn't be available.
 
Please run command "smc -k CHWA -r" to check.

SwiftWinds, 2025-09-01:
I did have AlDente and [Battery](https://github.com/actuallymentor/battery) at some point but uninstalled both of them in favor of this one, as your project allowed for me to make launch agents of when to start and stop the battery limiting (I enable it during work hours, as it's always plugged in during that time). Also, actuallymentor's Battery never worked for me.

I see:

```
pstree -p 78372
-+= 00001 root /sbin/launchd
 \-+= 01806 mateowan /opt/homebrew/opt/sleepwatcher/sbin/sleepwatcher -V -s /Users/mateowan/.sleep -w /Users/mateowan/.wakeup
   \-+- 78370 mateowan /bin/bash /Users/mateowan/.wakeup
     \--- 78372 root sudo smc -k CHWA -w 00
```

So I assumed that this project was the culprit as I believe it was this project that installed sleepwatcher.

SwiftWinds, 2025-09-01:
```
  if [[ $(smc -k CHWA -r) == _"no data"_ ]]; then
 chwa_has_data=false
 else
 chwa_has_data=true
 fi
 echo $chwa_has_data

true
```

I ran that command myself and indeed it says "no data" in stdout

js4jiang5, 2025-09-01:
> ```
> if [[ $(smc -k CHWA -r) == _"no data"_ ]]; then
>   chwa_has_data=false
> else
>   chwa_has_data=true
> fi
> echo $chwa_has_data
> 
> true
> ```
> 
> I ran that command myself and indeed it says "no data" in stdout

sorry, the markdown change the code. please run again
```
if [[ $(smc -k CHWA -r) == *"no data"* ]]; then
   chwa_has_data=false
else
   chwa_has_data=true
fi
echo $chwa_has_data
```

SwiftWinds, 2025-09-02:
```
if [[ $(smc -k CHWA -r) == *"no data"* ]]; then
   chwa_has_data=false
else
   chwa_has_data=true
fi
echo $chwa_has_data

false
```

SwiftWinds, 2025-09-02:
```
smc -k CHWA -w 00

Error: SMCWriteKey() = e00002bc
```

SwiftWinds, 2025-09-02:
OK thank you for the prompt help! :)

js4jiang5, 2025-09-02:
> OK thank you for the prompt help! :)

By the way, there's a buy in v2.0.25. Please run "battery update" to latest version v2.0.26.